prusa3d/PrusaSlicer #4365: "Travel," is driving me nuts

FlailAway, 2020-06-07:
### Version
2.2.0

### Operating system type + version
Mint 19 + Cinnamon

### 3D printer brand / version + firmware version (if known)
Anycubic Linear Delta, Marlin 1.1.19 bugfix

### Behavior
Weird jiggle with seam. No matter where I put it. The Jiggles is leaving a big Goober-like lump outside of the print, but is perfect on the inside. Is this a cruel joke? Kidding OK!!

[image]
[image]
[image]
[image]
[image]

rtyr, 2020-06-07:
I am not aware of any 3mf related to this, but I believe that "jiggle" is classic wipe movement. If you don't want it, disable it (wipe while retracting) or make it smaller (retract amount before wipe).

FlailAway, 2020-06-08:
Thanks, but the amount was already zero percent. The disable just made it smaller with 6 tiny moves and a snotty deposit doing it.

What about the wasted travel all over the Random Seam pic?

What's with all the to and fro Travel of the Rear Seam pic?

The Travel needs some serious optimization. I really do not like using Cura but for some prints it has 35% shorter times with none of these wasted Travels.

rtyr, 2020-06-08:
I believe you misunderstood that parameter, 0% retract amount before wipe means 100% wiping... I can't comment on snotty deposit on your printer, you are in the best position to fix your settings.

I am not sure what do you mean by wasted travel since you didn't provide 3mf. Cura is using the combing by default, which is basically a variant of "avoid crossing parameter" parameter in the PrusaSlicer. Avoid crossing parameter feature doesn't work idealy at the moment, it is a known issue mentioned in other issues.

Provide your 3MF.

FlailAway, 2020-06-13:
Finally back, been a bit busy, "Retired" means less time to relax! :)
New project, worse issue with Travel. I am still using PrusaSlicer (1:46 print print) as it prints a zillion times better than Cura (56m print time), but I do wish you could do something with the Travel as I am petty sure that's where all the time is going. Cura only traverses the posts when they are above the base-level.

[image]

and zip

[vent0613b.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4775410/vent0613b.3mf.zip)

rtyr, 2020-06-13:
Yes, this is a good example of what I described in my previous post. Cura is using the "combing feature" by default, which is a variant of "avoid crossing perimeters" feature found in other slicers. Using this feature, travel moves are usually longer, but without "unnecessary" retractions (probably the main time saver in your case) and without crossing the perimeters when possible. It all depends on particular configuration of the combing feature. But it is not always ideal, typical issue with the combing is missing material after long nonretracted travel. What we need to do is to improve "Avoid crossing perimeter" feature, we know it needs some rework. It is a known problem, see other related issues.

This is Cura without combing :-) (apples and apples).
[image]

FlailAway, 2020-06-14:
Thanks, but that is actually comparing good-apple with not-so-good-apple. It's like a race between a Pinto and a Lambo, when the race course is a mine-shaft. They both make the finish line together, but at the race-track ...

I seriously doubt the the difference is just Retractions when I have Wipe turned off as you had suggested previously and Retraction speed from memory was upped as well.

I have taken a short video that shows it doing multiple travel runs around the perimeter for every layer. It does a little dance at the two rounder corners. I can edit it down to just 3-layers so you can see what's happening during every single layer.

I will continue using PrusaSlicer for the better overall finish that is unrelated to time-to-print, but I do wish you could make those to be both good apples for Travel.

Please Close this if you wish.

FlailAway, 2020-06-14:
p.s.
I just turned off Retraction to see how much Retraction was hogging in both; PS saveed 15-minutes, Cura saved 8-minutes. "Huston, we have ..." :)

rtyr, 2020-06-14:
OK. I will not comment/explain this further, I believe I spent enough time with it. If you have something usefull to add, feel free to add it (gcode, layer height info, etc). I will leave this opened.

FlailAway, 2020-06-15:
OK, I'll keep checking back until it is fixed. I assume when you write, "it is a known issue mentioned in other issues." means it will be fixed -- eventually?

Is it really OK to say, "0% retract amount before wipe means 100% wiping?"

closed 2020-06-15 by FlailAway