prusa3d/PrusaSlicer #4367: Podpěry generované do vzduchu

joka-cz, 2020-06-08:
### Version
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,099MB
OpenGL installation
GL version:   4.3.0 - Build 10.18.15.4274
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4600
GLSL version: 4.30 - Build 10.18.15.4274

### MK3S FW 3.9.0

### Behavior
někdy se generují podpěry do vzduchu, je to vidět v přiloženém zazipovaném 3MF souboru. Na delší straně modelu je podpěra generovaná i u hrany modelu a je ve vzduchu. Na kratší straně modelu je podpěra generována, dle mého názoru, lépe a správně.

#### Project File (.3MF) where problem occurs
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/4744542/Test.zip)

bubnikv, 2021-02-19:
Nevidím problém

[image]

joka-cz, 2021-02-20:
v současné verzi už se nevyskytuje.
děkuji a zavírám.

closed 2021-02-20 by joka-cz