prusa3d/PrusaSlicer #4368: [Feature Request] Keep Cut visible when rotating or allow to rotating the cut

fscheps, 2020-06-08:
Today I was trying to cut a part of a piece and this cut had to be done in diagonal, it was tricky to try to get it right because:

a) The cut tool does not allow rotation.

b) I had to go to rotate tool and then come back to cut tool repetitively to "see" how the cut would end up looking like. 

Thanks for your consideration

lukasmatena, 2020-06-08:
Actually, this was already proposed internally and it will quite likely get attention in near future. I already gave it some thought and concluded that it may not be that difficult to implement.

fscheps, 2020-06-08:
Thanks a lot for the feedback!

Fernando

bttguy, 2021-06-16:
Any update on this FR?

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena