prusa3d/PrusaSlicer #4370: [Feature Request] Solid first layer for support material

MiggyMan, 2020-06-09:
**Version**

2.2.0

**Operating system type + version**

Win7 enterprise 

**3D printer brand / version + firmware version (if known)**

Ender 3 + Marlin 1.1.9.1

**Behavior**

When printing support material with increased desperation from the main print the support material tends to be prone to coming away from the build surface early on in the print, this could be alleviated by printing a solid base on the first layer for the support material. 

This is a new feature request

yoyo42, 2020-06-21:
I like this idea. I have some filaments that don't stick as well as others, they are fine when a small area is on the bed but the spiderweb you get at the base of supports can be too difficult, yet it's not really worth printing a whole raft to get around it. If it's implemented then this needs to interact well with brims and skirts.

rtyr, 2020-07-29:
Duplicate of #1165. Closing.

closed 2020-07-29 by rtyr