prusa3d/PrusaSlicer #4371: Incorrect first layer when Extrusion width is specified as 100%

0x3333, 2020-06-09:
### Version
2.2.0+ build 202003211132

### Mac OS Catalina 10.15.5

### Creality custom CR-20Pro

### Behavior
* Smaller extrusion width on the first layer.
* Set the same layer height to first and other layers(I used 0.25 for both), set in Print Settings, Advanced, Extrusion width, First Layer to 100%.
* The first layer should have a height equal to 0.25.
* Visually it is smaller.
  * [image]
  * [image]

rtyr, 2020-06-09:
It seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4169 to me. So expected behavior.

Extrusion width for 1st layer is calculated over first layer height if expressed as percentage. So 100% is 0.25mm extrusion width in your case. Can we close this issue?

[image]

0x3333, 2020-06-10:
Sure. My bad. Thanks.

closed 2020-06-10 by 0x3333