prusa3d/PrusaSlicer #4372: There isn't a .desktop file for Prusa Slicer

ericonr, 2020-06-09:
### Version

13a2e8e64da8eb05b1b64aef6c1e23785f5353ac

### Operating system type + version

Linux

### Behavior
* _Describe the problem_: no `PrusaSlicer.desktop` or `prusa-slicer.desktop` file.
* _Expected Results_: a desktop file to allow Prusa Slicer to be launched from a desktop environment's application launcher.

_Is this a new feature request?_: a bit, because the desktop file has to be parametrized for different places where the binary could be.

There are example files you could use as a starting point below, though an ideal one would have translations for the application name / description. The file should go in `/usr/share/applications` for distributed packages and `/usr/local/share/applications` for manual installs.

- https://aur.archlinux.org/cgit/aur.git/tree/prusa-slicer.desktop?h=prusa-slicer-git
- https://github.com/void-linux/void-packages/commit/a6ccc69dc48c5d96900504d3b213a933c0f4bd63#diff-be592f66502f233c561a955010493655

tamasmeszaros, 2020-06-10:
Hello @ericonr ,
so you are saying that we should bundle our own .desktop file in the source tree instead of bundling it with the appimage and let the distro package maintainers do it for us properly for each distro?

tamasmeszaros, 2020-06-11:
@ericonr I see your point.

tamasmeszaros, 2020-06-11:
A great feature would be an automatic AppImage generation with CMake, in which case the .desktop file would be an important component. I really don't see such a thing coming soon as no team member has the cycles for it, but a PR would be appreciated for sure.

ericonr, 2020-06-12:
I don't know enough CMake to help with the appimage creation, but I can try to look into the desktop files used for other similar applications to make a PR. If anyone sees this issue and wants to try their hand at it, feel free!

MooseOnTehLoose, 2024-11-10:
I just switched to Ubuntu 24.10 from Windows and found this thread, I also found a working example for anyone else that finds this:

1. Create /home/ USERNAME /Applications and move the .AppImage File there
    Don't use relative Paths, the desktop-file-install command doesn't like them.
    
2. Rename the AppImage to just PrusaSlicer.AppImage

3. Create a file in the same directory called PrusaSlicer.desktop:

```[Desktop Entry]
Name=PrusaSlicer
GenericName=PrusaSlicer
Type=Application
Categories=Engineering;Printing;
Exec=/home/tony/Applications/PrusaSlicer.AppImage
Icon=/home/tony/Applications/PrusaSlicer.jpg
Terminal=false
StartupNotify=true
```

4. Make PrusaSlicer.desktop executable:

`chmod a+x PrusaSlicer.desktop`

5. Move a PrusaSlicer.jpg icon into the same directory:

[image]

6. Install the .desktop file so it is available to search in the system:

`sudo desktop-file-install PrusaSlicer.desktop
`

You should now be able to search for the PrusaSlicer AppImage and pin it to your taskbar:

[image]

[image]

Happy Printing

Yohannfra, 2024-11-11:
@MooseOnTehLoose 

There is a small error in your PrusaSlicer.desktop, it needs to have 

```
[Desktop Entry]
```

At the top of the file, otherwise you will get this error from **desktop-file-install**

```
$ sudo desktop-file-install prusa-slicer.desktop
Error on file "prusa-slicer.desktop": Key file does not start with a group
```

MooseOnTehLoose, 2024-11-12:
@Yohannfra good catch! github formatting hid that for some reason when I added the code block (```[D ) and I didn't check. Its now fixed.

closed 2026-09-29 by prusaslicer-bot[bot]