prusa3d/PrusaSlicer #4380: Auto-dismiss unmount message after drive removal

SFPROXY, 2020-06-11:
### Version

PrusaSlicer-2.2.0+Fedora

### Operating system type + version

Linux 5.6.16-300.fc32.x86_64 x86_64

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S

### Behavior

Unmounting an SD card creates a message on if the unmount was succesful. Even when the card is removed, the message must be manually dismissed each time. This is a feature request; to auto-dismiss the dialogue box on drive removal or to disable the clunky message entirely.

#### Project File (.3MF) where problem occurs

N/A

lukasmatena, 2020-06-11:
@kocikdav is currently working on improving notifications emitted by PrusaSlicer and as far as I know, this change is part of the plan.

lukasmatena, 2020-08-05:
Implemented and in `master`. It will be in 2.3.0 release.
Closing.

closed 2020-08-05 by lukasmatena