prusa3d/PrusaSlicer #4381: Issue with computation of layers ?

bgiot, 2020-06-11:
### Version
2.2.0

### Operating system type + version
MacOS 10.15.5

### 3D printer brand / version + firmware version (if known)
MK3 with Bondtech extruder

### Behavior
Benchy model bottom part not slices correctly
PrusaSlicer defines only one layer before overhangs.  Other slicers like Ideamaker create 2 layers.  The text of the benchy is better with Ideamaker

#### Project File (.3MF) where problem occurs
[image]

neophyl, 2020-06-11:
Looks like you have the default 0.2mm elephants foot compensation on in Prusa Slicer which is making the text 'thicker' as its outlines are adjusted 0.2mm.  Try reprinting it with Elephants foot set to 0.

bgiot, 2020-06-11:
I set the elephants foot setting to zero.  the second layer is still the overhangs
[image]

lukasmatena, 2020-06-11:
Looking at the bed texture, the model is quite downscaled from the original benchy. Please enclose a 3MF. But if you scaled the model so the depth of the text is one layer, slicer will make it one layer. I don't know what you're comparing to.

bgiot, 2020-06-11:
I didn't change scale.  I just cut the benchy to only have the bottom part for my testings.
[3DBenchy_bottom.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4763606/3DBenchy_bottom.3mf.zip)

lukasmatena, 2020-06-11:
I don't see anything wrong. You are using layer height of 0.2mm, first layer is at 0.2mm, second at 0.4mm. The text is 0.3mm thick. Why should there be any text on second layer?

bgiot, 2020-06-11:
So, the software ignores - in a sense - half of a layer (0.1mm)... Indeed it's a big debate. "rounding issue" like

bgiot, 2020-06-11:
I think for bottom layers, the rounding must be changed. The current rounding is only valid for top layers I think... my two cents...

lukasmatena, 2020-06-11:
Yes, you're on the edge of the vertical resolution and choosing what is best to do is ambiguous. If you want more precision, lower your layer height. Even setting layer height to 0.15mm (default 0.15mm QUALITY profile) creates two layers of text in your case.
The math is a bit more complicated because the overhanging layer is assumed to be a bit higher then if squished against the layer below. It is not as simple as choosing between rounding up or down, as you probably think. Details like this may differ between different slicers.

bgiot, 2020-06-11:
Indeed...Adaptive layer height seems to handle it properly. I close this 'non' issue. Thanks!

closed 2020-06-11 by bgiot