prusa3d/PrusaSlicer #4386: The perimeters on the layer that contains a top solid infill are over extruded

dimonDDL, 2020-06-13:
### Version
2.2.0 win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
HyperCube 3D Printer/CNC by Tech2C
Ramps 1.4 + Marlin bugfix-2.0.x

### Behavior
Whenever a layer contains a top solid infill the outside perimeters become over extruded at that layer which can clearly be seen.
Printing the XYZCalibrationCube.stl at 0.2mm layer height and wipe enabled.
When the X starts printing (layer #18) the outside perimiter becomes over extruded.
Screenshot of the problematic area:
[image]
Printed in white is a file sliced in Cura while in Blue is Prusa Slicer 2.2.0
[image]

#### Project File (.3MF) where problem occurs
[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/4774913/xyzCalibration_cube.zip)

dimonDDL, 2020-06-14:
It appears that this is caused by the stock empty profile Prusa Slicer creates on install, since now I've obtained a profile that someone made for this specific printer which has a few settings altered that could have been triggering this behavior.

rtyr, 2020-06-14:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/991.

My result with your settings.

[image]

rtyr, 2020-06-14:
> 
> 
> It appears that this is caused by the stock empty profile Prusa Slicer creates on install, since now I've obtained a profile that someone made for this specific printer which has a few settings altered that could have been triggering this behavior.

It is so thick it almost looks like skipped step on Z axis. So it is solved with proper settings, can we close this?

dimonDDL, 2020-06-14:
> > It appears that this is caused by the stock empty profile Prusa Slicer creates on install, since now I've obtained a profile that someone made for this specific printer which has a few settings altered that could have been triggering this behavior.
> 
> It is so thick it almost looks like skipped step on Z axis. So is it solved with proper settings, can we close this?

Yes, I have narrowed it down to a not very stiff Z axis. However, it is a bit weird how Cura is somehow still managing a decent print with that problem. While with the broken Prusa Slicer gcode it makes this defect every time in the same spot.

closed 2020-06-14 by dimonDDL