prusa3d/PrusaSlicer #4389: Last used settings not saved

tonycstech, 2020-06-14:
Configuration snapshot needs to be taken automatically when program closes and load it automatically.
Every time i make change, close the program and come back, my changes are lost.

I do not want to create new presset for each small change i make.
I do not want to update current preset with small change i made.

tonycstech, 2020-06-14:
> It would be probably pretty confusing and unwanted behavior for most users.

That's how every slicer behaves. You need to use separate temporary
settings like what you already have as snapshot.

Your slicer is the only one that does this weird behavior and it asks you
of you want to exit without saving changes.

At the very least, it cold offer you option to save changes before exiting
but that'd the least logical thing to do.

Presets are untouched unless saved in the preset window, I don't see a
problem using temporary settings without effecting anything else and
without pointless warning messages when exit.

The thing is., When I export the model and start the print, I close the
program.
If print is doing something weird, i reopen the program and all the little
changes I did specifically for this print are reverted back to.preset i had
previously selected leaving me clueless about what I changed.

Problem with printing with wrong settings happens to me several times a day
because I close and reopen the program several times making changes such as
infill. After it's closed, I think that it's all as I left it and slice
another model. But once it starts printing, I realize it defaulted back to
present again.

Makes me sick caz I have to reslice again and restart the print.

neophyl, 2020-06-14:
You do know that you can import the settings from the gcode file you created don't you ?
File>Import>Import Config and then just select a gcode file that was made in Prusa Slicer.  Although that gets you only the settings and doesnt get you the model or any transforms/modifiers you placed.  It vcan be handy to check what settings you used for that gcode though.

The better option is to adjust your workflow to always save the file as a project.  Once you get in to the habit of doing that you can print and adjust very easily.  As everything is saved in a project file.

tonycstech, 2020-06-14:
Yes. But it's a time taking process .
Instead, program should automatically remember last settings.
Every slicer does it but yours.

In fact, I don't know any program that does reverts to some kind of default
every time it starts.
Even OS keeps what ever settings you chose.

I don't understand why you can't see it, its common sense.

tonycstech, 2020-06-14:
I don't want to save file as a project. I make changes to the file way too
often and keep printing them several times until they fit properly.
Creating project is nonsense.
This is not a 3d design application, this is slice and print.

neophyl, 2020-06-14:
I know several programs that revert to a default, Blender for instance. 

It’s just a workflow habit, I hit cntrl-s every time I click slice. What’s so onerous about saving it as a project ?  You can then come back whenever you want and change things. You iterate the part design even and reload from disk, slice, check the preview, if it looks good then you print. If something g doesn’t work then you load up the project and either adjust some slice settings or even load up your cad program and tweak the model if needed.  That way every thing is in sync. And if I do forget then there’s the already mentioned ability to load the settings in from the exported gcode. 

It’s open source you could always change it to work the way you want.

tonycstech, 2020-06-14:
I make changes to the model and don't always export it with the same name
so I could end up reimporting the file into a project.

It will also create a separate file for project and stl that I have to find
a place to keep which I don't want.
Easier to just export STL, slice it and delete it when done, keeping my
desktop clean and organized .
I don't even keep STL files. I export from my 3d editing programs, I have a
project file for 3d editing program. What ever I export is temporary untill
it's sliced.

Once model is sliced and exported to SD card, there is no need for the file
anymore.
So slicer project is completely useless feature for me.

Software is too complicated for me to make this change but for you it's a
matter of automatically saving snapshot and loading it at opening.

You also manage settings in too many different places where they don't
belong.
Printer should only have settings for bed size and number of nozzles.
Filament settings is a completely useless category because it effects how
it prints, for that reason everything to do with printing should be under
print settings.

I am not good at explaining my self in writing. If you were standing next
to me, you'd be like "yeah I see what you mean"

I get corporate software design directors come to me for feedback on a
tracking software at the hospital I work.
They came because none of my points make sense in writing but when they
came and I showed them how their software behaves and it's problems, they
were like "yeah I see it now"

I know how you think man, I make small programs my self in a different
language but not python so it's impossible for me to change anything.

I was back and forth with cratware developers a while back and they refused
to implement my suggestion.
They did implement one at the begining and it gave them success so much so,
they were bragging about it as if they never even heard of me.

Then they stopped listening to me all together and now their software is
completely abandoned.

Of all the things I can ask you to improve is: make print settings in one
place.
I have to switch 3 profiles when changing filament from petg to pla because
of how your settings are distributed

Have a good day.

Iqwertz, 2020-08-14:
I am completly on @tonycstech side! The current behavior is so annoying! It would improve the slicer so much if this small feature would be added!

closed 2026-09-29 by prusaslicer-bot[bot]