prusa3d/PrusaSlicer #4391: Start/Stop print at height [Feature request]

Mayoogh, 2020-06-15:
### Start and stop print at user-defined height.

**Start print heights**: Print starts at the height defined by the user. **Skirt/Birm/Raft** should be automatically disabled If Start print height is enabled 

**Stop print height**: Print stop at height defined by the user and home X and Y-axis to prevent damage print from hot nozzle

I did a quick search but couldn't find anything related to this other than the ColorPrint Prusa website
This feature is available using the [ColorPrint](https://www.prusaprinters.org/color-print/). It will be much better if it is integrated to PrusaSlicer.

rtyr, 2020-06-15:
> 
> **Start print heights**: Print starts at the height defined by the user. **Skirt/Birm/Raft** should be automatically disabled If Start print height is enabled

For what purpose? 

> **Stop print height**: Print stop at height defined by the user and home X and Y-axis to prevent damage print from hot nozzle
> This feature is available using the [ColorPrint](https://www.prusaprinters.org/color-print/). It will be much better if it is integrated to PrusaSlicer.

Already implemented.

[image]

foreachthing, 2020-06-15:
> ### Start and stop print at user-defined height.
> **Start print heights**: Print starts at the height defined by the user. **Skirt/Birm/Raft** should be automatically disabled If Start print height is enabled
> 

Is this to print on top of something?
Just use your favorite text editor and erase all the lines (after head-gcode) up to your desired Z-value.
Or, you could "add custom G-Code"-marker and then erase/mod/replace up to there with a custom post processing script.

Mayoogh, 2020-06-15:
> For what purpose?

Recovering failed print [Filament clog, power failure....]

Mayoogh, 2020-06-15:
> Is this to print on top of something?
> Just use your favorite text editor and erase all the lines (after head-gcode) up to your desired Z-value.
> Or, you could "add custom G-Code"-marker and then erase/mod/replace up to there with a custom post processing script.

I know it's possible that way. But It will be super simple to do that if there is an option like this 

[image]

pedjas, 2020-06-19:
I second this. It is reasonable and useful.

DioGreco, 2020-08-09:
This feature is very useful, they could be implemented in a furuta version. It is very bad to change the GCODE manually. I look forward to this resource.

fl0r1s, 2021-01-20:
You can also make a planar cut (default hotkey [C]) at a particular height and delete the part that is not required.

kiketurry, 2021-02-26:
@Mayoogh I also think it would be very useful.
I hope they end up adding it to a new version.

DaGwhite, 2021-08-08:
I use this feature all the time in Simplify3D.   I mostly use it starting from 0, and then end at 5 or 10 mm up to run test prints to check settings for finish & dimensional adjustments.  When it get to the set height, it stops and runs the usual ending script.

wolph, 2023-02-24:
> For what purpose?
>
>
> Already implemented.
>
> [image]

That feature is missing if you're using multiple extruders (MMU):

[image]

closed 2026-09-29 by prusaslicer-bot[bot]