prusa3d/PrusaSlicer #4392: [Bug?] Honeycomb doesn't connect to perimeter

alphapilot380, 2020-06-15:
### Version
Version 2.2.0+win64

### Operating system type + version
Win7 64bit

### 3D printer brand / version + firmware version (if known)
Own Build CoreXY kinematics

### Behavior
I'm trying to print this honeycomb panel which I designed myself, the single perimeter honeycomb is disconnected from the edges by PrusaSlicer. I tried fiddling with the settings but haven't found a solution so far. Possible bug? Other slicers don't have this behavior and so I'm suspecting a bug in PrusaSlicer.

[Honeycomb.zip](https://github.com/prusa3d/PrusaSlicer/files/4780425/Honeycomb.zip)
[image]
[image]

rtyr, 2020-06-15:
It is a side effect of "detect thin walls" parameter. Disable detect thin walls and change external perimeter extrusion width to 0.49 instead of 0.5, it should fix the issue.

alphapilot380, 2020-06-15:
Seems to have fixed it indeed... Printing in progress, thanks!
The 0.49mm perimeter extrusion width is because my honeycomb perimeters are 0.5mm or because I set the default extrusion width to 0.5?
I tried leaving it at 0.5mm with "detect thin walls" enabled but then my honeycomb partially disappears.

I don't fully understand the logic behind it yet unfortunately... Is this behavior by design or is there something that needs modification?

rtyr, 2020-06-15:
"Detect thin walls" parameter is basically a "last resort" option, which doesn't always work ideally at the moment (there are some opened issues related to this). 0.49 to fit the extrusion to 0.5mm honeycomb wall.

alphapilot380, 2020-06-15:
OK, got it. Thanks for the assistance!

closed 2020-06-15 by alphapilot380