prusa3d/PrusaSlicer #4393: PrusaSlicer pulls line through design first layer

Pinjosi, 2020-06-15:
### Version
2.2.0

Aplha

### Operating system type + version
Win10Pro

Sidewinder X1 Marlin Stock version

### Behavior
Pull at the first layer a line through the design
Just put in an STL and generate a Code

[image]

rtyr, 2020-06-15:
Please provide your gcode and project 3MF file.

Pinjosi, 2020-06-15:
i am bussy with this project now and the system did the same. One straight line through the first circle.
[Project 1.zip](https://github.com/prusa3d/PrusaSlicer/files/4782549/Project.1.zip)

rtyr, 2020-06-15:
Thanks. I previewed your gcode and there is no such line there, so your issue is most likely not PrusaSlicer related. Provided gcode is different object though.

rtyr, 2020-06-15:
I tried to print the gcode sliced from your project file. First layer is OK, no issues.

Pinjosi, 2020-06-15:
Wauw thats weird. I used a  profile with settings from 3d nexus. Could it be in this profile maybe?

Btw thanks for your effort so far. Very kind of you to help me out here.

Tomorrow i will post mine profile here also. If you have time to check maybe you can find something in there.

rtyr, 2020-06-15:
> Wauw thats weird. I used a profile with settings from 3d nexus. Could it be in this profile maybe?
> 
I don't think so, but we can check it. I would try different USB stick, corrupted file system can cause pretty weird things.

Pinjosi, 2020-06-15:
Okee Tomorrow i will try the other usb first.

Thanks again

Pinjosi, 2020-06-16:
With n new usb same line cross the design. Going to check the 3d nexus now

rtyr, 2020-06-20:
Can we close this issue?

Pinjosi, 2020-06-20:
Yes you can close this. I can not find the issue. Switching back to Cura
infortunatally

closed 2020-06-20 by rtyr