prusa3d/PrusaSlicer #4400: PrusaSlicer 2.3.0 zip archive contains 2.2.0

gaellafond, 2020-06-18:
### Version
2.3.0

### Operating system type + version
Linux (Ubuntu)

### Behavior
The zip archive https://www.prusa3d.com/downloads/drivers/prusa3d_linux_2_3_0.zip contains PrusaSlicer-**2.2.0**+linux-x64-202003211856.AppImage instead of version **2.3.0**

neophyl, 2020-06-18:
The PACKAGE version (which contains other things besides Prusa Slicer) happens to be 2.3.0.  Its unfortunate that this version number is so close to the Slicer software version to cause this confusion but its quite common in software releases to version the packages themselves too.  
Suffice it to say that the versions are unrelated.

gaellafond, 2020-06-18:
Thank you Neophyl

The "zip" archive only contains the PrusaSlicer, so the version number has no "excuse" to not match :D

What is even more confusing is that PrusaSlicer 2.3.0 alpha is out (source code at least).

Is the 2.3.0 appImage available anywhere? Any idea when it will be available?

neophyl, 2020-06-18:
Didnt realise the linux version only had slicer in it.  The windows one has other stuff too.  I'm pretty sure they keep the package versions in sync with each other, so theres the excuse lol.

As for an actual  2.3.0 I'm not brave enough to compile it myself so if it isnt on the github release page then I dont really want to use an alpha that the developers dont think is at a stage to build yet.  Of course as soon as it is available I'll give it a whirl.

gaellafond, 2020-06-18:
I watched this YouTube video where Josef Prusa is talking about all the cool new features of 2.3.0
https://youtu.be/NAW2JR-D_QU

So I headed directly to the PrusaSlicer website and there was a 2.3.0 download link! I was excited to try the new features, so I downloaded it and it and the Zip archive only contained version 2.2.0, the exact same file that has been available for months already...

I was quite quite disappointed

Thanks for the explanation. I guess the real issue is that Prusa is mixing their product PrusaSlicer with other stuff and make it available on a website dedicated to PrusaSlicer using a version number unrelated to the product, but which is almost the same. It would make more sense to completely separate PrusaSlicer from the other stuff, since the website is dedicated to PrusaSlicer.

I will close the issue since it was not really a bug, it was just a case of misleading labelling...

closed 2020-06-18 by gaellafond