prusa3d/PrusaSlicer #4402: Feature suggestion for developers of Prusa Slicer.

odidprusauser, 2020-06-18:
### Version I would like to request to put into Prusa Slicer (on the Plater Page at bottom to right of Feature type)- a check box located to the right of the Legend checkbox a box labeled Temperature Legend that would show the temperature of material as printed per layer displayed in different colors for each temperature.  This would be very helpful for example when making a temperature tower.  I just finished putting together some custom temperature towers and the temperature tower object had different temperatures written in gradients of 5 degrees per segment.  I thought that I changed the temperature in Printer Settings Custom G-Code as well as the Filament Settings First Layer and other layers Temperature appropriately.  In some I didn’t either the First Layer temp or Other layer temps or failed to put the right temperatures for the Layers in the Custom G-Code.  I printed one of the temperature towers as an example and left it to print after a few layers.  After the print, I noticed it didn’t look as I expected, so I reprinted it and watched it as it reprinted which took a few hours.  During the reprinting I observed that was printed everything at the wrong temperature with no gradient.  If their was a temperature legend checkbox that could be toggled on or off, It would have been useful to toggle this on and I would have expected to see the temperature tower in a rainbow of colors showing each segment of temperature that would displayed in a different color.  But in an instant would  be able to visually see that as printed there was no temperature change as all the segments would have been one color showing no temperature changes.  I would have known immediately what the error was,  rather than have to sit for hours watching the temperature as it prints that I have made an error and needed to go back and fix the settings for the first temperature layer, other layers and the custom G-CODE.  I think that adding a toggled temperature legend to show the as printed temperatures to be displayed in different colors would be a desired feature for Prusa Slicer for a future version.

Boby71, 2020-11-22:
I guess it will be hard to parse the custom gcode for the visualization platter, but maybe this could be the solution?
https://github.com/prusa3d/PrusaSlicer/issues/5230

In combination with your request to visualize the different temps it would make very much sense together...

bubnikv, 2020-11-23:
> I guess it will be hard to parse the custom gcode for the visualization platter, b

I believe we already do.

grostim, 2021-06-13:
I don't see this feature in prusa slicer 2.3.1 (macos) . Am i missing it ?

neophyl, 2021-06-13:
No you aren't missing it.  The preview doesn't do temperature as an option in the drop down.  

I suspect the comment about we already do was in response to Boby71 where he says it will be hard to parse the custom gcode.  In fact if you read the post it quoted that section in the reply.

Sineos, 2021-06-13:
https://github.com/prusa3d/PrusaSlicer/commit/c9666ea257e4161bdcf5591e91cca2724f562228

Seems planned for 2.4.0

bubnikv, 2021-07-29:
Implemented by #6106 thanks @combolek.
It will be part of 2.4.0-alpha1.

closed 2021-07-29 by bubnikv