prusa3d/PrusaSlicer #4403: [featurerequest] Delaying hot end heating

pedjas, 2020-06-19:
### Version
2.2

### Behavior
PrusaSlicer starts heating both hot end and bed at the same time. Hot end gets to target temperature much quicker and cooks filament while waiting for bed to heat up.

With filaments that require higher temperatures this issue is even more evident - hot end cooks filament on higher temperature and waits longer for bed to heat up.

It would be good to have defaults set in a manner that hot end heating starts later, when bed is already at required temperature or close enough.

Even when both temperatures are reached there is still delay waiting for temperature to spread evenly on bed (which is a must for successful print of large objects).

If bed is heated first then time waiting for hot end to heat up would also serve as time for spreading temperature evenly on bed, reducing additional delay.

Users can achieve this by altering configuration in PrusaSlicer but it is advisable to have this settings as default.

_Is this a new feature request?_
yes

rtyr, 2020-07-17:
I don't think that such change is needed. It is up to the author of the particular printer profile.

pedjas, 2020-07-26:
> I don't think that such change is needed. It is up to the author of the particular printer profile.

That is exactly the point. to have this default for Pruša printers.

rtyr, 2021-02-22:
We will not change it for now. Closing.

closed 2021-02-22 by rtyr