prusa3d/PrusaSlicer #4409: Z-Lift 0 setting ignored in 2.2.0

NothingButGunpowder, 2020-06-20:
### Version
2.2.0

### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
i3 MK3S, firmware 3.8.1-2869

### Behavior
When setting Z-lift (under Printer Settings -> Extruder -> Retract settings) to 0 the setting is ignored.  It should prevent z-lift between retract and movement, but in 2.2.0 the printer continues to perform a lift before moving.

#### Project File (.3MF) where problem occurs
This is not specific to a project and happens in any file I create.

rtyr, 2020-06-20:
You are probably using filament profile with some retraction overrides.

[image]

NothingButGunpowder, 2020-06-20:
I double checked and no overrides are set in filament overrides.

NothingButGunpowder, 2020-06-20:
I am updating firmware to 3.9 now, just in case that's the issue.

NothingButGunpowder, 2020-06-21:
The new firmware seems to fix the issue.

closed 2020-06-21 by NothingButGunpowder