prusa3d/PrusaSlicer #4410: [REQ] Navigation cube to snap to camera view

mvasilakis, 2020-06-20:
### Version
2.2.0+

### Operating system type + version
MacOS 10.15.5
### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Feature Request

A navigation cube similar to what Autodesk uses that allows for a click to snap to view. (Top, Right, Bottom...)
[image]

neophyl, 2020-06-21:
Just use the keyboard shortcuts for the views.

AndKe, 2021-01-25:
The current navigation is .. not good, especially when zoomed in on small parts on a large printer.

Please adapt's  FreeCAD , the default and navigation options - those would make all happy, and work well regardless of zoom.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena