prusa3d/PrusaSlicer #4415: Resolving mDNS hostnames fails often

hoppke, 2020-06-21:
### Version
2.2.0+win64

### Operating system type + version
Windows 10, latest (build 2004)

mDNS resolution done by Win10's built-in "dns client" service (instead of the classic Apple Bonjour service)

### Behavior
Submitting sliced gcode to Octoprint on local network fails for me most of the time due to DNS timeouts.
[image]

If I retry enough times "send to printer" will finally succeed
[image]

Ideally I'd like to see slicer wait a few seconds longer before giving up. I've not checked the code, but I'd hope this particular timeout is still something slicer can control.

I have a workaround, all it takes is changing the slicer config to use the Pi's IP instead of mdns name, but of course mdns is more convenient.

### "Back seat developer" comment
I think I only started seeing this issue when I moved away from Apple mDNS stack (Bonjour service) and allowed the native MS dns client to take over as the mDNS resolver in my Win10. It does its job well in other places (ssh clients, ping, browsers - I can sometimes observe a few seconds delay, but it works - so I hope this could be just a matter of tweaking some name resolution or socket connection timeout in Slicer?

bubnikv, 2020-12-03:
I think Windows only implements
https://en.wikipedia.org/wiki/Zero-configuration_networking#NetBIOS_Service_Discovery
it does not implement the Apple's own protocol.

I think your issue is the same as
https://github.com/prusa3d/PrusaSlicer/issues/1781

gnydick, 2024-08-24:
Why isn't this using the system resolver? Seems silly to implement DNS directly

closed 2026-09-29 by prusaslicer-bot[bot]