prusa3d/PrusaSlicer #4416: Bad seam position and bad quality on Perimeters with PrusaSlicer.

BTopbas, 2020-06-21:
### Version
2.2.0+

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Creality CR-10 / Klipper

### Behavior
Perimeter quality is bad because of seams. 

I sliced the same file with S3d and after Prusaslicer. I calibrated every setting. And in PrusaSlicer seam position setting is aligned same as s3d.

[image]
[image]

Here are pictures. The bottom part sliced with s3d and seam aligned one single line. Top part sliced with PrusaSlicer and you will see seam is everywhere.

BTopbas, 2020-06-21:
PS: When I changed seam position to rear problem solved. Only a single line center of the rear created.

thierryzoller, 2020-06-24:
The S*D seam is also not really consistent..

bubnikv, 2022-06-29:
Solved by the user by selecting a rear seam.
Also the upcoming PrusaSlicer 2.5.0-alpha2 will align seams better.