prusa3d/PrusaSlicer #4417: Feature Request - Mouse Scroll Zoom

altuor, 2020-06-21:
### Version
2.2.0

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
MK3
### Behavior
* When zooming (mouse scroll button) PrusaSlicer Zooms towards the center of the 3D object zone
* In many tools tools (ie Fusion 360) the zoom function homes in on the mouse cursor instead, which is very handy to target a specific zone, adjusting while zooming. It would be really great if PrusaSlicer could do that too
Feature Request

phantomkingx, 2020-10-29:
This would be particularly useful when trying to inspect periphery sections of large models.  My current workaround is to place a dummy modifier object where I want to zoom on the model and pressing 'z' while the dummy modifier is selected.

bubnikv, 2020-12-12:
There is now a "reverse direction of zoom with mouse wheel" checkbox in preferences.
Available in PrusaSlicer 2.3.0-beta2.

foreachthing, 2020-12-12:
@bubnikv that is not the same. The OP wants to zoom to the cursor, not reverse-wheel-zoom. 
Currently, I think, PS zooms to center of screen (which I find annoying as well, because my CAD zooms to cursor) .

bubnikv, 2020-12-12:
thanks, I was reading too fast.

closed 2026-09-29 by prusaslicer-bot[bot]