prusa3d/PrusaSlicer #4419: Impossible to configure the printer

BlopperBbz, 2020-06-21:
### Version
Version 2.2.0+win64

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
The first launch of PrusaSlicer is not working each time the software is executed. 
A second launch is openning an empty error popup then openning the interface. 
All configuration attempt (configuration wizard of the printer) is failing ending with a non responding task.

As looking in windows process, the first launch is using constently around 30% of processor and is never ending. 
The second launch is kind of "unlocking" the first one.

I tried launching in trace mode in order to find the source, the task is blocked at following location :
>\>prusa-slicer-console.exe --loglevel=9
[2020-06-21 20:23:51.942796] [0x00000528] [info]    Checking if indices need to be installed from resources...
[2020-06-21 20:23:51.945812] [0x00000528] [trace]   System language detected (user locales and such): fr_FR
[2020-06-21 20:23:51.946785] [0x00000528] [trace]   Best translation language detected (may be different from user locales): fr_FR
[2020-06-21 20:23:51.946785] [0x00000528] [trace]   Switching wxLocales to fr_FR 

I tried removing the AppData folder, the software complete reinstallation, changing installation location without any success. 

Any Guess of a way to proceed ?

lukasmatena, 2020-06-23:
Thanks for the log. The logging is not too extensive at that point so it is not possible to tell where exactly it hangs, but last output says that it is about to switch locales. If you can run PS in some way, can you try switching it to English to rule out problems caused by switching to French? If you can't do it, you can add  `translation_language = en_US` key into first block of PrusaSlicer.ini. You can also try to switch your system to English (probably after creating a new user for that to be safe - I'm not a Windows guy so I'd be cautious), that should force PS to run in English.

And I better ask: Is this a build from our buildserver or have you compiled it yourself? Has it ever worked for you (older versions, etc)? If so, was there anything special happening before the problem started appearing? Is there anything special about your system so we can try to reproduce it ourselves?

BlopperBbz, 2020-06-27:
Ok, so the languague switch didn't affect the result. Wierd stuff, there is no PrusaSlicer.ini in my installation folder (windows install maybe), I had to switch the system language. Anyhow the log is quite the same 
> .\prusa-slicer-console.exe --loglevel=9
[2020-06-27 16:44:55.281702] [0x000020b8] [info]    Checking if indices need to be installed from resources...
[2020-06-27 16:44:55.298657] [0x000020b8] [trace]   System language detected (user locales and such): en_US
[2020-06-27 16:44:55.299654] [0x000020b8] [trace]   Best translation language detected (may be different from user locales): en_GB
[2020-06-27 16:44:55.299654] [0x000020b8] [trace]   Switching wxLocales to en_GB

And for the version, this is the one directly downloaded from the prusaslicer page (2.2.0 from https://www.prusa3d.fr/prusaslicer/ , yes french again ^^ ). I never tried any other version in the past as I'm quite new to 3d Printing. The same installer is working on a other computer also in win10 without any problems.

For the custom part in the OS, the only specificity that I made was a mklink performed for the c:\Users folder
>PS C:\> dir -force | ?{$_.LinkType} | select FullName,LinkType,Target
>FullName               LinkType Target
>--------               -------- ------
>C:\Program Files       Junction {d:\Program Files}
C:\Program Files (x86) Junction {d:\Program Files (x86)}
C:\ProgramData         Junction {d:\ProgramData}
C:\Users               Junction {e:\Users}

Because of that specificity I also tried to re-install the PrusaSlicer directly on D and used a relative user folder next to the executable:
> .\prusa-slicer-console.exe --datadir Profile
and same result.

closed 2026-09-29 by prusaslicer-bot[bot]