prusa3d/PrusaSlicer #4420: Plater graphics area not displayed

jdog3000, 2020-06-22:
### Version
_Version of PrusaSlicer used goes here_
2.0.0+win64-201905201652

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #3603 
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
2.0.0+win64-201905201652

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa I3 Mk2

### Behavior
* _Describe the problem_
Plater graphics area is not shown.  No model, no bed volume, no controls.  Only the sidebar containing the print settings and object names is displayed.
* _Steps needed to reproduce the problem_
Start the app.  Doesn't matter if I load a model or not.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

rtyr, 2021-02-22:
Is this issue still valid? Does it happen with 2.3.0?

rtyr, 2023-09-12:
No response. I am closing the issue.

closed 2023-09-12 by rtyr