prusa3d/PrusaSlicer #4422: [FR] Import and finish-up "Thin wall improvements (v3)" from upstream Slic3r

VanessaE, 2020-06-22:
### Version

git commit 91897589928789b261ca0dc735ffd46f2b0b99f2

### Operating system type + version

Debian

### 3D printer brand / version + firmware version (if known)

Custom Hypercube running Marlin `bugfix-2.0.x` (at commit 8000182b).

### Behavior

For the longest time now, thin wall behavior on Slic3r and Prusaslicer has been of sub-standard quality - bends at the ends of lines, over-extrusion, sometimes just not being drawn, and a few other things.

See https://github.com/slic3r/Slic3r/pull/4534 followed https://github.com/slic3r/Slic3r/pull/4790 for more details on the existing code's behavior, and discussion on how fixes were being implemented into Slic3r (it worked great, but has not been finished/merged).

@supermerill  would you consider porting your Slic3r code to Prusaslicer?

#### Project File (.3MF) where problem occurs

(N/A)

supermerill, 2020-06-24:
Sure, it's not that hard,
It's basically the changes from the files in the pr, but from my current fork master branch.

Their is some things why i didn't did it yet:
 - I can still discover some sub-optimal behavior, so it evolve a bit over time.
 - I don't want to bother and waste prusa time a third time on this issue, until my "solution" is almost perfect, as the previous improvements weren't good enough. So the goal was the pr merge into the upstream before submitting again.
 - I didn't merged the reforged PerimeterGenerator from prusa. Have to do that before creating the pr.

VanessaE, 2020-09-20:
@supermerill any movement on this?

supermerill, 2020-09-20:
I have to finish the upstream one first.
I also added a little code to merge the thin walls into the perimete loop, to avoid oozing and retractions. Need some more time to perfect it.

Didn't had the time yet to merge prusa's perimetergenerator changes. Maybe on the next big merge.

If someone else want to submit a pr here, you can do it. It's basically the new class medialaxis.cpp and "some" changes in perimetergenerator.

closed 2026-09-29 by prusaslicer-bot[bot]