prusa3d/PrusaSlicer #4423: Progress (bar) stuck; endless generating g-code

foreachthing, 2020-06-22:
### Version
2.3.0a

https://github.com/prusa3d/PrusaSlicer/commit/d6e040c282a9992d4cfebe85b3a931a43b2773d0

### Operating system type + version
Win 10 x64

### Behavior
* Progress bar is stuck and the end 
[image]
   - Taskmanager `PrusaSlicer-2.3.0-alpha0+unknown...` working at 14-16% at 260 MB ram. CPU time currently at 19 minutes.
* Cannot export g-code
 - Slic3r exports gcode within seconds

#### Project File (.3MF) where problem occurs
3mf and config:
[Battery_Adapter_AA-C.zip](https://github.com/prusa3d/PrusaSlicer/files/4813267/Battery_Adapter_AA-C.zip)

lukasmatena, 2020-06-23:
I just tried it the exact same git revision and your 3MF sliced in less than a minute (Linux, i7). You are using 'avoid crossing perimeters' and other features from Print Settings->Layers and Perimeters->Quality, which are known to have very detrimental effects on slicing performance, but your model is not that complex. I don't know what's the problem.

Have you attached exactly the 3MF that leads to the problem?

foreachthing, 2020-06-23:
> Have you attached exactly the 3MF that leads to the problem?

Yes. Exactly that 3mf. Freshly zipped and still warm ;-)

--- hours later ---

Yeah, it was the "avoid crossing perimeter" *) setting. Disabling it helps the slicing time. But I just **hate it**, when the nozzle crosses parts! _(it has already knocked over some other parts! Those very tall and had a small cross-section)_ 
And also, when printing tubes/holes, I get a lot of stringing. Retraction is already at 6 mm.

*) _it slows down gcode generating_! Yeah, no sh..!

lukasmatena, 2020-06-23:
Avoid crossing perimeters is known to be slow and it will be rewritten for the better one day. It is still beyond me why it would slice for 19+ minutes though. I could imagine that with complex models, but not with this one.

I'm marking this with 'path planning' and will change the title a bit if you don't mind, so it can be found with the other issues related to avoid crossing perimeters feature.

> it has already knocked over some other parts

Your z-lift is set to 0 mm, that might be the place to improve it. Don't forget that avoid crossing perimeters is not equivalent to 'never cross perimeters'. It just makes it somewhat less frequent, so it will not solve your problem.

foreachthing, 2020-06-23:
I guess, I can close here. This is probably a "won't fix".

The result. too, is not very .... useful!
Avoid perimeter true:
[image]

Avoid perimeter false:
[image]

XOR:
[image]
Not worth the trouble ...

closed 2020-06-23 by foreachthing

lukasmatena, 2020-06-23:
> This is probably a "won't fix".

It is not. The fact that avoid crossing perimeters is slow AND even useless in some cases is not a "won't fix". The fix will probably involve rewriting that function completely. We will get to it one day. But there are several different issues that track it so this one may as well stay closed.

foreachthing, 2020-06-23:
@lukasmatena, yeah, I'm sure someone can and will fix this. Maybe not in _this_ issue; that's what I meant. Sorry, that I didn't express myself clear enough.