prusa3d/PrusaSlicer #4425: Monotonous infill for middle solid fill layers

vintagepc, 2020-06-22:
### Version
2.3.0-Alpha

### Operating system type + version
Linux (OpenSuSE 42.3 x64)

### 3D printer brand / version + firmware version (if known)
MK3S/MMU2S 3.9.0/1.0.6

### Behavior
Please make Monotonous infill also able to be applied to middle solid infill layers. It leads to some improvement but much of this is lost if the layer underneath still exhibits the scarring/ridges generated by standard rectilinear fill. 

_Is this a new feature request?_
Sort of - it's an improvement on a new feature. 

#### Project File (.3MF) where problem occurs
N/A - any non-rectangular part that would result in ridges with standard rectilinear fill.

timur-tabi, 2020-10-17:
For the record, "monotonous" is not the right word. The correct word is "monotonic".

n8bot, 2020-10-17:
https://github.com/prusa3d/PrusaSlicer/commit/7fbdda90803fc2653e2bddc6217ef641acb35578

vintagepc, 2020-10-17:
Fair enough, I was going by the explicit nomenclature at the time.

edwardyeeks, 2020-10-19:
FWIW, I too notice a scar or two from the solid fill layer directly beneath the top fill that's set to monotonic. Besides providing settings to set solid infill as monotonic, another option would be providing settings where one can set the number of top fill layers. I think setting the top 2 or 3 layers as monotonic should be enough to hide away any scars from the rectilinear solid fills beneath them.

bubnikv, 2020-10-20:
Just adding the Monotonic infill to the list of existing "Fill pattern"
does not cut it, as it changes the sparse infill as well.

We would need to add another config option, either "solid fill pattern", or
a "solid fill monotonous" checkbox,

or just use the Monotonic infill for all solid infill if the top infill is
set to Monotonic (very simple and likely passable, as the Monotonic infill
works as good as the rectilinear, possibly slower to optimize)
or just use the Monotonic infill for couple of layers below (more complex
to implement, possibly slower to slice, possibly problematic as it would
split the solid infill to montonic and regular part.

mjticoat, 2020-12-13:
If the extrusion width is chosen not too wide for the top layers, it gives me nicer results in a single pass in PrusaSlicer than the Ironing feature. But indeed, any mess from bumps in the lower layers are shown through the nice monotonic later.
The algorithm is already there, please allow the Monotonic pattern to be applied to all solid layers. Then the top finish should be flawless.

iso250, 2020-12-13:
Is there any word on whether this is being worked on?  The scarring on the second from top layer when printing with a 0.6mm nozzle (what I use 90% of the time) shows through the top monotonic layer and looks so very ugly.

It seems odd to add monotonic top fill to make prints look pretty whilst leaving the problem with the solid fill, which ruins the top monotonic fill.

5teveO, 2020-12-17:
Just to add my support for this.. monotonic solid infill layers would be excellent. Having monotonic infill as the last layer before ironing is not enough to hide the traditional infill scars underneath it before ironing. I dont care about the few minute extra - i prefer clean prints!

Steve

stepan-martinek, 2021-01-04:
Is there any reason why anyone would want non-monotonic infills? Speed concerns? Don't think so. The difference in time is negligible in contrast with the quality improvement. I think that all solid infills should be always monotonic. Period.

bubnikv, 2021-01-06:
> Is there any word on whether this is being worked on?

It is not being worked on, because we need to roll out the current release, thus we are busy with fixing issues and processing the communication storm on github and other social networks.

> The algorithm is already there, please allow the Monotonic pattern to be applied to all solid layers. Then the top finish should be flawless.

It is too late in the release cycle. We need to decide 
1) When to enable it
2) How will it influence the print. 

Currently we have the Monotonous infill enabled by default in our profiles, thus if we enable Montonous infill for all solid infills, we need to make sure, that it will behave.

swissbear, 2021-02-07:
I printed lots of thin sheets (1mm thick, 5 layers) in transparent petg. The monotonic top layer looks great, but the 3 layers inside dont look nice at all and the bumps and scars are easily visible. 
My vote also goes for "all solid infills are always monotonic". its a bit more calculation power, but the benefit is way superior.

bubnikv, 2021-06-03:
In PrusaSlicer 2.4.0-alpha1, if you have the Montonic infill pattern set for top surfaces, all internal solid infill will be monotonic. We have it that way for some time already and we did not experience any issue.

closed 2021-06-03 by bubnikv

iso250, 2021-06-03:
Hey there.

Are you able to link to the alpha?  I can't find it.

Many thanks,

Lehane.

vintagepc, 2021-06-03:
It's probably not ready yet, you will need to build it yourself until they finish working on it. 

Thanks! this should be a significant print quality improvement especially at lower layer heights where this tends to "show through" much more.

iso250, 2021-06-03:
Funny enough I’ve found the problem to be much worse when using large
nozzles, 0.6mm and bigger. With a 1.0mm nozzle it looks horrendous!!!!

Some prints I really need a perfectly smooth top. For that I have to use
super slicer as it has monotonic solid infill. I can’t wait until
PrusaSlicer has it as I really don’t like super slicer. That thing just
behaves poorly and isn’t as reliable or consistent as PrusaSlicer.

bubnikv, 2021-06-03:
@iso250 
> For that I have to use super slicer as it has monotonic solid infill.

This is lovely, because I am the author of Monotonic infill, so it was in PrusaSlicer before SuperSlicer merged it.

iso250, 2021-06-03:
Awesome work! I think monotonic infill is the biggest and most important
feature added to PrusaSlicer in a very long time!!! 😀

vintagepc, 2021-06-03:
well, until the whole paint-mmu-colours feature is released :rofl: that'll be quite something.

iso250, 2021-06-03:
> well, until the whole paint-mmu-colours feature is released 🤣 that'll be

That sounds good, but that’s an update that benefits a subset of people.
Monotonic benefits everyone!!!

iso250, 2021-06-05:
Vojtěch, you mentioned that in the alpha monotonic solid infill is enabled
when monotonic top infill is enabled.

Is it possible that monotonic solid infill be a separate selection that we
can choose in the slicer settings?  Otherwise if we want to use a different
top infill pattern we won't get the benefit of monotonic solid infill.  For
example, I really like concentric and use it a lot.  If selecting
concentric top fill doesn't add monotonic solid infill then I'll get the
scars showing up through my beautiful concentric top layer!!