prusa3d/PrusaSlicer #4426: [Feature Request] MMU Select filament for first layer

brettvitaz, 2020-06-22:
### Version
2.2.0 (any)

### Operating system type + version
Mac OS 15.5

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2s

### Behavior
I would like to select the filament used for the first layer, instead of always being filament #1. 

_Is this a new feature request?_
Yes

rtyr, 2020-06-22:
Can you elaborate? Maybe include 3MF project and screenshot. I am not sure what do you mean.

brettvitaz, 2020-06-22:
For example I would like to assign Filament 2 to start the print. There is no option to "Change Extruder" on Layer 1. [image]

You can see the "Change Extruder" icon on Layer 2 and above.
[image] 

[Shape-Cylinder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4815412/Shape-Cylinder.3mf.zip)

rtyr, 2020-06-22:
I see. It can be done, see the screenshot.

[image]

Area you highlighted is meant for color/extruder changes. There is no reason to do color/extruder change on the first layer. First layer is printed with selected extruder as shown on my screenshot.

brettvitaz, 2020-06-22:
Thank you, that was the missing link for me. 

As I was experimenting around the "Change Extruder" area, I found that there is a small window where the mouse can access the extruder selection. Maybe this is a potential problem then?
[image]

brettvitaz, 2020-06-23:
My question has been answered. Closing.

closed 2020-06-23 by brettvitaz