prusa3d/PrusaSlicer #4428: "Collapse right panel" button does not appear until application is started with the panel hidden.

n8bot, 2020-06-22:
### Version
2.3.0-alpha0 built from tm_mini_supports branch as at: https://github.com/prusa3d/PrusaSlicer/tree/351c0133e8caab0ed0b8eb47adaacf3b47690938

### Operating system type + version
Win7

### Behavior
The icon/button with the symbol that looks like "<<>>," and the tooltip "Collapse right panel," does not appear until the application has been started once with the panel hidden.

Afterwards, the button always seems to appear, whether the panel was left hidden or shown at last shutdown.

I now have the button appearing, but it was never there for the past few days when using the build. It wasn't until I discovered the menu item View->Collapse sidebar, that I realized this functionality was in the build. It wasn't until I started a second instance of the application when that sidebar was hidden that the icon button appeared for me.

As I can't reproduce this behaviour myself, since the button now always appears, I suggest building a fresh copy of the version I stated above, and see if that button is there upon first start.

lukasmatena, 2020-06-23:
Whether that button is visible or not is decided in Options->Preferences. The menu item View->Collapse sidebar just collapses sidebar, but does not enable the config option. That only happens if the GUI is recreated with the sidebar hidden. I'm not sure if it's the intended behaviour. @YuSanka could probably tell.

n8bot, 2020-06-24:
Interesting. I now see that the option is enabled in preferences. I never noticed it. Is it set as enabled by default?

YuSanka, 2020-06-24:
Hi, this looks like a bug in the application. The option to show the "expand/collapse" button is automatically turned on if the application is started with the right panel being collapsed.
We'll fix this so that the state of the option to show/hide the expand/collapse button doesn't change on application restart.
This bug will be fixed in the next release. Thanks for your report.

closed 2020-06-25 by YuSanka