prusa3d/PrusaSlicer #4429: Printer will not work

PhilipPuttifer, 2020-06-23:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-06-23:
> Printer will not work

I'm sorry, we need more info than this...

PhilipPuttifer, 2020-06-23:
Thank you for reply. The printer does all the operations of changing filament and goes through the bel level checks with no problem. On trying to print anything all is well except nothing is printed. But all seems fine until that point. Filament loads and unloads as normal. Printer seems normal .Just will not print at all from any source. But all appears normal and workingPlease help. Thanks from PhilipSent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/prusa3d/PrusaSlicer/issues/4429#issuecomment-648186245",
"url": "https://github.com/prusa3d/PrusaSlicer/issues/4429#issuecomment-648186245",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

PhilipPuttifer, 2020-06-23:
I have just tried to print a well printed item and all that appears on the print bed is what I would describe as random dust. An outline appears lightly printed but the rest looks like dust. I do not know but hope it helps. PhilipSent from my Samsung Galaxy smartphone.

rtyr, 2020-06-24:
This is most likely not PrusaSlicer related problem. I asked our customer support to find out, what printer do you have. They informed me, that you already contacted them about this problem. I am closing this issue, tech-support will handle it.

closed 2020-06-24 by rtyr

PhilipPuttifer, 2020-06-24:
Lukas,I am very sorry for not replying to you direct. It was rude of me and I apologise. It was only that I had been in communication with  Douglas several times. The problem l have with my printer is that all appears well except that when trying to print all goes well with bed level tests etc, but no filament is deposited on the bed. The only time filament appears is when it is changed and am asked if it correct colour is shown.Can you please help?Once again I apologise for not replying to you directly.Pjilip.Sent from my Samsung Galaxy smartphone.

lukasmatena, 2020-06-25:
> I am very sorry for not replying to you direct. It was rude of me 

No, it wasn't. Replying in a thread is completely fine and polite. But I agree with @rtyr that your issue seems not to be PrusaSlicer related. We as PrusaSlicer developers have neither time nor knowledge to take care of it. Contacting support (as you apparently already have done) is a way to go. I hope they help you find a way to fix the issue.

lukasmatena, 2020-06-25:
> Can you provide advice to repair the printer?

Sadly, no. You must contact tech-support. This issue tracker is only for PrusaSlicer developers, we cannot provide support for anything else.