prusa3d/PrusaSlicer #4430: PrusaSlicer PAINFULLY slowly in windows 10 Pro 64 bit

pyrow14, 2020-06-23:
### Version
2.2.0

### Operating system type + version
win 10 Pro 64 bit
Build version 2004 and 1909

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with MMU2s running 3.9.0 (1.0.6) from May

### Behavior
unresponsive performance all around
Double click to open software > problem occurs throughout the entire software experience 

#### Project File (.3MF) where problem occurs
All projects

eceforge, 2020-12-31:
So I've found this to be the case when the slicer is open full-screen on a high-resolution monitor (I happen to have 4k). If I reduce the window to a quarter-screen, performance picks up quite a bit. It's kinda odd, because the exact same computer hardware booted to linux works wonderfully fine at full screen ( I think this might count as the only time a graphics-related thing has worked better in linux😄 ).

bubnikv, 2021-01-04:
PLease test with PrusaSlicer 2.3.0-rc2.

HairBears, 2021-03-14:
Issue is still there. I'm currently using PrusaSlicer-2.3.0+win64-202101111315 on a 4k monitor. In fullscreen every tab switch takes 2 seconds while in a small windowed mode it responds immediately.

closed 2026-09-29 by prusaslicer-bot[bot]