prusa3d/PrusaSlicer #4432: Manual colour change - nozzle goes to last place on previous layer before starting new layer.

UltraSalem, 2020-06-24:
### Version
2.2.0 linux x64

### Operating system type + version
Linux - Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Genuine Prusa i3 mk3s

### Behavior
When using the manual colour change at layer function (the + on the layer preview slider), after loading the new filament colour the nozzle returns to the last place on the previous layer before moving to the first place on the new layer. This can leave a messy minor stringing of the new colour.

The nozzle should instead move directly to the first place on the new layer after filament change, without returning to the previous position.

sw3de, 2020-07-02:
This is also true when you have virtual nozzles, after each color change it starts by touching the last printed point and then moves on to the new printing area.
I have assumed this was a firmware issue related to the M600 command? But if it’s not or if it is possible to correct with Prusa Slicer it would be very nice.

sw3de, 2020-07-02:
Oh, and a suggestion if it is a firmware issue, as a workaround you could add a move right before the M600 to the starting point of the next area but +1mm in Z axis.

fscheps, 2020-08-23:
It would be quite useful if the filament could be primmed too, as you just changed filament to ensure it comes out neat and tidy.

rkarlsba, 2020-12-23:
I can vouch for this. I don't have an MMU, but I use multicolour prints sometimes for labels etc and I've seen this happen there. Usually not a big issue, but it may leave a spot that can sometimes be hard to remove.

fscheps, 2020-12-23:
Also once you confirmed that the color change was done correctly, the nozzle still extrudes like 3-4 cm of filament that shouldn’t happen, I have to “catch it on the fly” with a tool so it doesn’t ruin the print.

bubnikv, 2021-01-06:
This sounds more like a firmware feature, however it may not be as simple as it seems. 

The firmware may likely safely drop the first travel move after the color change event. It may drop all travel moves after the color change event, but what if some "avoid crossing perimeters" feature is active, producing complex travel moves?

rtyr, 2021-02-19:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2672.

DeanDavis, 2021-11-10:
This is actually quite a big problem for some jobs. I have a job where my build plate is filled with multiple instances and after the color change it drags a filament line across the entire job. The next layer then prints over this line since it is pretty impossible to remove during the print.

Depending on the layers this introduces a line in the prints that shouldn't be there.

Ozbones, 2022-08-15:
I was trying to print two instances of a tall skinny tower with multiple filament changes and getting constant failures. The filament change would leave a blob on the second tower then move to the first tower and begin printing with the new colour. By the time the new layer is printed on the first instance the blob left by the colour change has hardened causing the extruder to crash though the now solid filament.
The print would always fail at the same point. The second tower would come loose and always after the filament change at 120mm high which was several hours into the print. The extruder would crash trhough the blob left by the filament change then catch on the print and pull it far enough for the print to fail.

I fixed my problem by editing the gcode and moving the M600 filament change command  down several lines. Instead of doing the filament change then moving to the next print position
[image]
I moved to the next position then changed colour
[image]

I still get a blob when doing the filament change but it's located where the printer is about to start printing the same colour it's about to print. It's also often printed over and stread out while still warm.

The result is not a single print failure.

smtpboy, 2023-07-06:
I'll pile on this one as well, that little "blob" is really problematic on prints where you may only have a few layers before a color change happens.  I print a lot of challenge coin and raised logo designs and that little blob is enough to knock extrusions that are very small right off the build plate, ruining the print.  As others have pointed out, if instead of returning to its previous position, if the printhead would go to the new print point, it would eliminate a lot of this problem and likely just smear the blob when it starts printing again, and it would at least be the right color.  If that's not doable, perhaps just doing a very small retraction right before it returns to its origin point might also solve this.  But the "extruding while in motion" after the color change is really a challenge.

arielbernal, 2023-11-16:
This being around since 2020, I guess we can say that this is not longer a bug.  Just add this feature to the documentation and you're done  :)

lukasmatena, 2024-02-18:
Solved in 2.7.2-alpha2. Closing.

closed 2024-02-18 by lukasmatena