prusa3d/PrusaSlicer #4433: Flashforge Creator 3 - Flashforge Dreamer Firmware

flash370, 2020-06-24:
Version
2.2.0

Operating system type + version
Any

3D printer brand / version + firmware version (if known)
Flashforge Creator 3 - Flashforge Dreamer Firmware 

Add firmware for this printer
Expetially in Duplicate Mode, for generate commented gcode to permit the printer build automatically the Raft under the higher extruder.
Feature Request

mishave, 2021-06-10:
Second this...

wmmc88, 2021-06-10:
Since FF keeps a lot of their stuff closed source, it'd take a good amount of effort to figure out what exactly needs to be generated. Also it might be easier to write a post processing script to do this instead of modifying prusaslicer directly

bubnikv, 2023-02-24:
This feature has to be provided by an external contributor and it will likely not happen.
Closing.

closed 2023-02-24 by bubnikv