prusa3d/PrusaSlicer #4436: [Feature Request] Compute printed part center of mass

josepher44, 2020-06-24:
### Version
2.2

### Operating system type + version
Any/Windows 10

### 3D printer brand / version + firmware version (if known)
Any

### Behavior

It would be incredibly useful for applications involving balanced spinning objects if the printed part's center of mass location and moment of inertia data could be computed for the actual sliced model. Some specific elements which would be useful:
*Center of mass shown both with numerical XYZ position and a glyph in preview mode
*Comparison to values computed from the isotropic STL
*Ability to exclude certain filaments, or certain feature types, for example soluble supports, from this calculation

_Is this a new feature request?_ 

Yes

remram44, 2026-09-02:
Maybe you should wait for the new release to be out of alpha before closing issues wholesale. I'm going to assume this issue still exists, though I won't be able to test until you actually release...

philip-soerensen, 2026-09-02:
The 3.0.0-alpha11 does have an option to show center of mass, so it seems like your wish was granted!

I believe it's only shown visually, but if you want the actual numerical location, perhaps it could be extracted with a plugin.