prusa3d/PrusaSlicer #4438: [Feature Request] Objects and modifiers opacity

voxel3, 2020-06-24:
### Version
2.2.0+win64

### Operating system type + version
Windows 7 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3. Stock
Firmware 3.8.1-2869

### Behavior
* _Describe the problem_
It is not posible to see modifiers placed inside objects.

[image]

* _Expected Results_
The opacity of the objects could be definable.

[image]

* _Actual Results_
Placing parts or modifiers inside objects is dificult because the main object is opaque.
When a modifier is not selected, it becomes semitransparent but not the main part.

### Feature Request
* _Add an option to set the objects opacity_
(Or at least a global xray mode)

Similar to the "object printable property" (the eye icon on the property panel), could be an opacity property column to set the transparency of parts and modifiers.
This would facilitate the placement of modifiers when they are completely within objects.

[image]

neophyl, 2020-06-24:
#3678 I already requested basically this.  Without the fancy pictures though :)

voxel3, 2020-06-25:
Oops, you are right.
I looked for transparency but found nothing.
I added X-ray mode at the last minute in case it's easier to implement, though there is currently support for semi-transparency per object (modifiers are semi-transparent when not selected).
Perhaps making that control accessible to the user may be a good possibility.

itrmc, 2021-04-13:
I'm 100% behind this, and would add that modifier object should have an option for opacity control. Objects which become opaque when deselected obscure items the build plate, including the modified object. When trying to align a modifier, it is always helpful to see the item being modified.

NVGG-creations, 2021-05-25:
I would add my support for this.  It is a little odd that we use modifier primitives to change settings on specific parts of our models without being able to see the parts that are being affected.

Jonny98731, 2021-11-19:
I am 100% with you - this would be a very important feature!

bcorbett, 2022-03-17:
+1 
Trying to add a modifier and align it with an interior part of the model. I'm having to guess on the position, slice, then guess again, over and over.

Gone4Dirt, 2022-04-29:
+1 on this feature request.  Would be very useful.

guido66611x, 2023-08-31:
+1 on this.  I often have to make parts that fit in other parts (e.g. storage box), but can't see inside.  What I do now is to leave both enabled (printable), slice it, move the slider to look through the slices, move the parts by guessing where they should go, reslice, etc., then turn off (not printable) the inner part at the end.

Being able to enable transparency, or adjust opacity, would save me so much time!

WhiteDragonD, 2024-02-17:
+50

kubispe1, 2024-10-29:
A new view option would be fantastic and highly beneficial. I wouldn’t complicate it with additional settings for each volume. A standard view mode, like wireframe or shaded view with hidden edges, would be a valuable addition to the 3D editor viewer—similar to the viewing options in CAD and other software. This approach aligns well with best practices.
[image]
I'm closing this issue as it duplicates #3678 and the current #13534, which are linked to this one.
Thank you for your detailed description and comments.

closed 2024-10-29 by kubispe1