prusa3d/PrusaSlicer #4440: Per-layer preview inspection of vase mode model skips all but first and top solid bottom layers

fiveangle, 2020-06-25:
- PS 2.2.0+
- MacOS 10.15.5
- MacBookPro11,3 w/dual graphics GForce GT 750 + Intel Iris Pro
- MK3 3.9.0

* Slice vase mode model with large number of solid bottom layers
* Inspect per-layer in Preview mode starting from first layer moving up
* Observe rendering skip all middle layers and go directly to the last (top) solid bottom layer immediately prior to start of single-perimeter vase mode

Video showing behavior:
https://youtu.be/hZi1Vmk_l20

3mf example:
[Draining_eToothbrush_holder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4828466/Draining_eToothbrush_holder.3mf.zip)

bubnikv, 2020-12-09:
I am not sure I understand the problem. Would you please retest with PrusaSlicer 2.3.0-beta2 and possibly with the beta3, which will fix some vase mode issues?

fiveangle, 2020-12-20:
Same behavior (as shown in the video). 

Unlike Simplify3D "Factory" files, it seems the PS 3MF files does not retain print settings (which begs the question of what the use of them is for ?) but instead simply saves the currently selected print profile.  This means that one has to save the profile with that print's specific configuration in order to capture it.  I was not aware of this, so I apologize for that.

So, complete steps required to see issue:

- Load example 3MF
- Set bottom layers to "10"
- Set spiral vase mode
- Slice
- In preview window, click and hold "top" current layer selection widget and drag widget upward as slow as possible
- Expected: Preview moves from first layer to next layer, and so on
- Observe: Preview skip from bottom-most bottom solid layer to top-most bottom solid layer, then continue with correct preview of continuous spiral vase mode.

fiveangle, 2020-12-20:
I did some more poking, and it seems the issue is the mouse scaling changes from some arbitrary number of layers when sliding the "current layer" handle widget when on the bottom solid layers, then scales to a "normal" scaling once the "current layer" is above the "bottom solid layers" section.  This is still unexpected and confusing to the user since no mater how small a movement the mouse makes, the preview cannot go below a very large increment.

Work-around is to use "up"/"down" scroll button on multi-button mouse (if you have one) and preview correctly increments one layer at a time.

bubnikv, 2021-01-05:
PrusaSlicer 2.3.0-rc2 is out https://github.com/prusa3d/PrusaSlicer/releases
with reworked G-code viewer for spiral vase mode.

fiveangle, 2021-01-06:
2.3.0-RC2 layer viewer shows no change from 2.2.0... still jumps from preview of first solid bottom layer to topmost solid bottom layer, then increments as expected through each increment of the vasemode section above the solid layers.

[image]

bubnikv, 2021-01-07:
I cannot reproduce it.
For a 3 layer bottom and spiral vase, the slider iterates over all the
three bottom layers.

1st layer

[image: image.png]

2nd layer

[image: image.png]

3rd layer

[image: image.png]

start of the spiral vase, 1st segment

[image: image.png]

fiveangle, 2021-01-07:
Yes, I observe the same with just 3 layers.  Is there a problem with the recipe I provided ?

> Set bottom layers to "10"

bubnikv, 2021-01-08:
The layers are there, but they are not accessible when sliding the slider with mouse, the vertical resolution is not sufficient. However if you iterate over the layers using the cursor up / down keys or with the mouse scroll wheel, the layers are iterated correctly.

fiveangle, 2021-01-08:
> The layers are there, but they are not accessible when sliding the slider with mouse, the vertical resolution is not sufficient.

Your statement doesn't correlate with the observations.  The issue is not due to a hardware limitation of mouse resolution or even the software slider resolution.  If you re-read my report above, you'll see I mentioned that above the bottom solid layers section, the mouse movements iterate every layer with the slider, even though the mouse resolution hardware is unchanged.

fiveangle, 2021-12-23:
Bug still present in Prusaslicer 2.4.0

fiveangle, 2021-12-23:
> The layers are there, but they are not accessible when sliding the slider with mouse, the vertical resolution is not sufficient. However if y accessible when sliding the slider with mouse, the vertical resolution is not sufficient. However if you iterate over the layers using the cursor up / down keys or with the mouse scroll wheel, the layers are iterated correctly.

When slicing normally, the progression works as expected so it is not a vertical resolution limitation of the mouse. However, because every segment in the vase-mode section is essentially a psuedo "new layer" is there an upper limit to the maximum number of "layers" that the UI can scroll over ?  If so, it could be that the non-vase-mode section at the bottom (the "Bottom Layers" number of layers) scrolls up many layers because the UI treats that `Bottom Layers` section just as it does each vase mode arc segment, so skips many layers based on the linear division of the maximum UT verticle scroll units (a term I just made up) ?  That could explain the behavior. If so, would it be possible to treat the non-vase-mode layers at the bottom as normal layers then divide whatever is left over and use that value for the bulk scroll iterator for the vase mode section ?

For example, if maximum "vertical mouse scroll units" is 32768, `Bottom Layers` is 9 and number of vase-mode-segments is 100,000, current behavior seems to be to make minimum vertical scroll units ( `Bottom Layers` + `vase mode segments` ) / `32768`  which would be 100,009 / 32768  or ~3 layers as minimum layers to skip for every vertical mouse scroll movement.

When slicing normally, the progression works as expected so it is not a vertical resolution limit of the mouse. However, because every segment in the vase-mode section is essentially a psuedo "new layer" is there an upper limit to the number of "layers" that the UI can scroll over ?  If so, it could be that the non-vase-mode section at the bottom (the `Bottom Layers` number of layers) scrolls up many layers because the UI treats that `Bottom Layers` just as it does each spiral segment, so skips many layers based on the linear division of the UI vertical slider ?  That could explain the behavior. If so, would it be possible to treat the non-vase-mode layers as normal layers then divide whatever is left over and use that value for the bulk scroll iterator for the vase mode section ?

For example, if maximum "vertical mouse scroll units" is 10,000, `Bottom Layers` is 10 and number of vase-mode-segments is 99,9990, current behavior seems to be to make minimum vertical scroll units:

 ( `Bottom Layers` + `vase mode segments` ) / `10,000`

which would be:

100,000 / 10,000 = 10

Which means any upward movement of the mouse from the base layer would result in the next visual layer showing as layer 10, then the next render is where the vase mode section starts so renders 10 additional segments of the vasemode section, and so on...

What should happen is the `Bottom Layers` should be consumed from the "vertical mouse scroll units" so every layer gets displayed with every mouse movement (or at least identical to how is done when not printing in vase mode), then when the vase mode begins, the segments should be displayed in jump of ("vertical mouse scroll units" - `Bottom Layers`) / 10,000

Does this make any sense ?  If someone could point me to the code where the UI scroll data is determined I might be able to explain in the exact code values rather than my confusing psuedocode.

bubnikv, 2021-12-23:
> Unlike Simplify3D "Factory" files, it seems the PS 3MF files does not retain print settings

This is not true. PrusaSlicer's 3MFs contain the complete profiles.

bubnikv, 2021-12-23:
The issue here is that the vertical scroll bar resolution is limited by the
number of screen pixels with regard to mouse control. If there are 5x more
pseudo layers than there are screen pixels along the vertical scroll bar,
only every 5th layer is selected by the mouse. If one controls the scroll
bar with a keyboard up / down buttons, every layer is accessible.

The situation is not ideal, we need to implement a special mode for the
vase mode G-code visualization, with the vertical scroll bar jumping over
the original layers while the horizontal scroll bar would iterate over the
segments along a single turn of the spiral. However this mode will take
time and effort to implement and it will make the code more complex.

bubnikv, 2021-12-23:
Also the horizontal scroll bar legend is broken in PrusaSlicer 2.4.0 for
vase mode.

[image: image.png]

fiveangle, 2021-12-30:
> > Unlike Simplify3D "Factory" files, it seems the PS 3MF files does not retain print settings
> 
> This is not true. PrusaSlicer's 3MFs contain the complete profiles.

I dug in this more long before and found that if the current profile is changed (unsaved), the changes are thrown away and only the saved profile is retained in the 3MF.  I will retry with the current PS 2.4 to see if this behavior has been fixed/changed because I think it has. My previous observation was many PS iterations ago, but it definitely didn't work... I setup the print to show the problem, saved the 3MF, opened it, and the running (unsaved profile) settings were not retained.

> If there are 5x more pseudo layers than there are screen pixels along the vertical scroll bar, only every 5th layer is selected by the mouse.

Yes, that is what I was alluding to above.  Okay, so fixing the visual unexpected jumping of base layers when previewing vasemode prints will require segmenting the scroll wheel behavior so that the vase mode segments are treated with a different ratio than the standard layers.  Understood.

fiveangle, 2021-12-30:
@bubnikv - is there a tag you have for lower priority things like "visual artifact" or something so this item can be prioritized accordingly, but not forgotten ?

bubnikv, 2022-01-04:
No, but I will ask @enricoturri1966 to put it on his list.

bubnikv, 2022-01-27:
Fixed with 7ed80e0a8d354ea30d67e8e5cef156b22643f92f

The fix will be released with PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv