prusa3d/PrusaSlicer #4442: Overhang perimeters are treated as bridges instead of a kind of perimeter

gnydick, 2020-06-25:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro 64-bit  Version 2004 build 19041.329

### Feature Request
Overhangs that your printer is capable of shouldn't be considered bridges in the speed calculation. Bridges are completely unsupported and require anywhere from 30-60 mm/sec so as not to sag.

On the other hand, overhang perimeters often do better with slower speeds so they can be cooled in place without curling up.

I think a new speed settings should be added

- overhang external perimeters
- overhang perimeters
- overhang small perimeters

faboaic, 2020-08-08:
Even worse... bridging perimeters (as well as bridges) are printed with a giant layer height of nozzle diameter.
E.g. 0.4mm nozzle and 0.2mm layerheight producing bridges and bridging perimeters of a totally ugly 0.4mm.

This leads to worst first layer on support, too. 

I have disabled "detect bridging perimeters" so that support works fine but bridging is not so good with this setup. :-(

halfshavedyaks, 2021-09-06:
+1 on this

a bridge is supported/connected at both ends. if it isn't supported at both ends it is an overhang not a bridge

an overhang is supported by an adjacent perimeter, and may or may not be connected at the ends

they need to be treated totally separately.

also I'm seeing overhangs that would be unprintable without supports detected as bridges, and the speed increased, even though support is generated under them.

see #5938 - a closely related topic

closed 2026-09-29 by prusaslicer-bot[bot]