prusa3d/PrusaSlicer #4443: Allow Configuration Snapshot Custom Path

swissfreek, 2020-06-25:
### Version
2.2.0

### Operating system type + version
macOS Catalina

### Behavior
Currently on macOS the configuration snapshots are stored in a subfolder in the Application Support directory. This is fine except that that folder is not part of the iCloud Drive backup process, so anytime the system gets reinstalled (which happened to me at the Apple Store recently even though I asked them not to), you will lose your carefully crafted profile unless you know in advance and remember to back up that folder manually. It's like the ONE thing on my system that doesn't get stored in iCloud Drive by default that causes me a total pain in the butt when I have to set it all up from scratch again.

It would be super handy if there were a way to specify the location of the configuration snapshot folder in the preferences, so that you could, for example, put it in the Documents folder, and then it would automatically be saved to iCloud Drive and backed up with everything else.

This is a new feature request.

neophyl, 2020-06-25:
Why don’t you regularly use the create config bundle option ?  It exports all the customised print, filament and printer profiles to a file that can be imported back in to any other instance of the software. It’s how to easily copy stuff between alpha/beta/releases, work/home setups and also into/from other forks like Super Slicer. 

Also I vaguely recall someone else asking about this and the reps she was to use some Mac folder type mapping/linking I think. I dont have a Mac so can’t recall the specifics.

swissfreek, 2020-06-25:
> Why don’t you regularly use the create config bundle option ? 

Easy: because I forget 😜

> Also I vaguely recall someone else asking about this and the reps she was to use some Mac folder type mapping/linking I think. I dont have a Mac so can’t recall the specifics.

Yeah, I was googling to see if there was a way I could add that specific folder to the iCloud Drive upload, but it doesn't look like I can. If I was using Time Machine for backups, then I know I could add it, but I'm currently without a "permanent" desk so I don't have an external drive to use that is regularly plugged in.

lukasmatena, 2020-06-26:
I'm not an Apple guy, but isn't the `Application Support` directory exactly the location where Apple gods want everyone to store application configuration (akin to AppData on Windows and .config on Linux)? If they decided to just nuke it without your consent or if they somehow actively prevent you from backing that folder up, I would complain at Apple or start using different OS. I don't feel that we should workaround everything, we would not do anything else.

All that being said, starting PrusaSlicer from command line with `--datadir` parameter should allow you to reconfigure the default configuration directory. For example `./prusa-slicer --datadir=/home/username/apple_allows_me_to_backup_this_one/prusaslicerconfig`. Set up some shortcut that will always run it with this parameter (unless that is forbidden too) and you should be good to go.

I'm sorry for my tone in case I misunderstood the question.

swissfreek, 2020-06-26:
> I'm not an Apple guy, but isn't the `Application Support` directory exactly the location where Apple gods want everyone to store application configuration (akin to AppData on Windows and .config on [...]

No offense taken, I understand what you mean. When I say it was nuked, what I meant was my system was reinstalled without warning by a service tech and I lost all my user data that wasn’t stored on iCloud, which is very little, but my Prusa Settings was one of those things.

I’m a nerd, but not a developer, so I’m not really sure what the rules are as far as where Apple wants apps to store data, but I can say that many apps have distinct folders in the iCloud Drive where they store user data (specifically only user data, like files and documents created by the user not placed there by the app when freshly installed) and it automatically gets backed up. This may be because of apps that are installed by iOS, I don’t know. Again, I’m a nerd but not a developer, so it’s very possible (even very likely) that I’m the one who misunderstands.

I guess I will make a More concerted effort to remember to periodically export my config to a folder that is backed up. Probably wouldn’t have to do it very often to at least have a decent baseline to start from if this were to happen again.

lukasmatena, 2020-06-26:
@swissfreek Like I said, I'm not an Apple guy (and I highly doubt I ever will be). All that I'm saying is that PrusaSlicer configuration cannot be in a more suitable place. Whether it gets backed up by default is another story, and stories written by Apple are usually incomprehensible to me. It is quite possible that it does not just backup whole Application Support directory, but requires the application to do some kind of registration first. We can try to look at it.

Until then, you know where the folder is and now you can change its location. Convincing macOS that it is worth backing up is up to you at this point.

swissfreek, 2020-06-26:
@lukasmatena Understood thank you. I think configuration Snapshot is possibly the wrong term. Basically I was just hoping that custom profiles or the config bundle or what have you might be put somewhere that it can be captured by iCloud’s backup system. For sure Application Support is not part of that, but you are right, that is probably where the snapshots belong.

What you changed the issue name to is probably more accurate, I guess what I’m looking for is iCloud backup support for user files (config bundle). Not a critical feature, just would be nice to have for the forgetful people like me 😜

kenthinson, 2020-07-19:
@swissfreek Why not just create a symbolic link to a folder in the iCloud Drive folder? Then when Prusia Slicer saves into the symbolic link the files are redirected.

swissfreek, 2020-07-19:
@kenthinson Actually yeah that looks like it would do what I want it to. Thanks, I learned something new today!

koisdigala, 2021-01-09:
Be sure to test whether the symlink actually works. All backup programs have a option or policy on whether symbolic links (or whatever your OS/FS calls it) are followed.  Two undesireable things which often happen:
- symlink directory is copied but not followed, so the backed up folder is empty
- symlink is completely ignored

Speaking of ignoring, you can ignore me if you already tested it and that's what you meant by "looks like it would do what I want it to".

jordanbrown0, 2024-05-15:
Exactly the same problem applies on Windows with OneDrive.  PrusaSlicer puts the data in AppData, and OneDrive doesn't back up AppData.

closed 2026-09-29 by prusaslicer-bot[bot]