prusa3d/PrusaSlicer #4445: IlmBase::Half can not be found!

Megidd, 2020-06-27:
### Version

On `master` branch:

```
$ git describe
version_2.3.0-alpha0-193-g990bfc0d7
```

### Operating system type + version

Windows 10 Pro
Version 1903

### Behavior

While trying to build with CMake, this error is received:

```
CMake Error at cmake/modules/FindOpenVDB.cmake:322 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:346 (just_fail)
  C:/Users/m3/repos/vcpkg/scripts/buildsystems/vcpkg.cmake:331 (_find_package)
  CMakeLists.txt:444 (find_package)
```

I have already installed Ilmbase and all the other dependencies by vcpkg:

```
>.\vcpkg.exe install IlmBase:x64-windows
```

Megidd, 2020-06-27:
### Possible cause

Might be related to `OpenEXR` being entangled with `Ilmbase`, not sure!

Megidd, 2020-06-27:
Setting this CMake option explicitly didn't help:

```
-DILMBASE_ROOT=C:\Users\m3\repos\vcpkg\installed\x64-windows
```

Megidd, 2020-06-27:
Also this CMake option didn't help:

```
-DILMBASE_INCLUDE_DIRS=C:\Users\m3\repos\vcpkg\installed\x64-windows\include\OpenEXR
```

tamasmeszaros, 2020-06-30:
We don't support vcpkg. Please follow the official [instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md) to build the dependencies and PrusaSlicer.

closed 2020-10-30 by lukasmatena