prusa3d/PrusaSlicer #4446: What is "browsing mode"?

BirkBinnard, 2020-06-27:
### Version
2.20

### Operating system type + version
Windows 10/64 Home current version

### 3D printer brand / version + firmware version (if known)
FLSun QQS-Pro with Ver. 2.5 firmware

### Behavior
The first solid layer that is printed on top of infill is very sparse and causes my extruder to click incessantly. The second such layer is better - no extruder clicking - but is still quite sparse. The 3rd layer is almost solid. The 4th & 5th layers are solid.

I've posted this problem with photos on the PrusaSlicer Facebook group. This is the link to the post:

[https://www.facebook.com/groups/2144493972517241/?post_id=2386800918286544&comment_id=2386814964951806&notif_id=1593280362932057&notif_t=group_comment&ref=notif](url)

Unfortunately you have to log in to Facebook to see it. This is the first response I got:

The first layer on top of the infill is in browsing mode so the passes are wider with bridge settings. But overall it’s normal that it takes at least 4 layers to get a good finish on top. You can also increase the infill density to reduce the spacing between the lines.

I've never heard of "browsing mode" before so don't know if this is something real or not. But if it is, is there a way to disable it?

BirkBinnard, 2020-06-28:
Problem caused by bad STL file - corrected by 3D Builder.

lukasmatena, 2020-06-28:
Closing - caused by a bad model.

> PS: It would be nice if github replaced this forum software with one that allows file & image uploads. Someone needs to tell the admins that disk space is cheap these days.

GitHub in fact does support file and image uploads. Drag and drop your file into the text box and all should be fine.

closed 2020-06-28 by lukasmatena