prusa3d/PrusaSlicer #4447: [ENHANCEMENT] Multi-document support

scooper4711, 2020-06-27:
### Version
2.2

### Operating system type + version
_Mac OS X 10.13

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
New Feature request.
I would like the ability to have multiple project files open at once, each in a separate window. In this way, I can do things such as compare the how various settings impact the slicing on the same model, I can have one project "in progress" where I'm working on support blockers and enforcers while also being able to spin up a quick slice and print. etc etc.
The idea is Multi Document Interface, similar to how MS Word or many, many other programs function. The "Project file" in this case is the document.

_Is this a new feature request?_
Yes

n8bot, 2020-06-28:
I do not know how mac-os works, but on Windows, you can start multiple copies of PrusaSlicer. I have three open right now.

scooper4711, 2020-06-28:
Mac -not so much.

n8bot, 2020-06-29:
It's three entirely separate instances of the program. The RAM usage seems appropriate. One instance of the program without anything loaded takes about 48 MB of ram. An instance of the program with a very detailed preview open takes 3 GB. Another instance with a very basic preview open takes 190 MB.

Is there no way to run multiple instances of an application on macos?

scooper4711, 2020-06-29:
You can do some command-line magic as a work-around:
````open -n /Applications/Original\ Prusa\ Drivers/PrusaSlicer.app````
But, my primary slicer is an older laptop which I keep next to the printer. It doesn't have a lot of memory to spare. Would love it if PrusaSlicer supported the MDI approach.
Also...not sure how well two instances will play together with regards to file locks, preference saves etc. E.g. What happens if one instance changes a named setting that the other instance is using?

bubnikv, 2020-10-12:
PrusaSlicer will allow opening new instances in the upcoming 2.3-alpha1 from the menu (Window->Open new instance) and from the context menu of the docker icon.

We do not think there is any memory issue with that approach as the image of the program is shared among instances and the base program memory is quite low. Most of the memory is allocated by the models, slicing, undo/redo and mainly the G-code preview, therefore starting new instance for each slicer consumes just a tiny bit more than it would if the multiple instances were running in a single process.

Closing.

bubnikv, 2020-10-12:
We are quite aware that the approach of starting multiple instances of the same application is neither recommended nor supported by the OSX framework. Namely, the Finder is only aware of a single instance, therefore drag & dropping a file over the application icon in Finder or in Docker sends the "Open file" event to some random, likely the first instance of PrusaSlicer. Other than that, our multi-instance implementation in PrusaSlicer 2.3 is quite usable.

closed 2020-10-12 by bubnikv

rkarlsba, 2020-12-20:
I've missed this functionality for a while, and it coming to 2.3 is great. Testing now, with PrusaSlicer 2.3.0-rc1, that menu choice is greyed out. Any idea how to enable it?

lukasmatena, 2020-12-20:
@rkarlsba It has to be allowed in Preferences (Configuration -> Preferences). There was an issue with that (the logic was inverted, #5463), but it should be fixed in 2.3.0-rc1. Does it solve your problem?