prusa3d/PrusaSlicer #4448: Vendor settings stuck at version 1.1.2 on Linux

goncalossilva, 2020-06-28:
### Version
2.2.0

### Operating system type + version
Fedora 32
(and macOS 10.14.6)

### 3D printer brand / version + firmware version (if known)
MK3S, stock firmware 3.9.0, no modifications.

### Behavior
The printer and filament settings on Linux (AppImage) are for version 1.1.2. The ones on macOS are for version 1.1.6 (which I assume is the latest). Somehow the Linux version is bundling outdated versions.

Here's a screenshot of Linux vs Mac on the system printer page (same thing applies for filament and print system settings):

[image]

[image]

Notice the vendor version disparity.

I have also tried deleting the `~/.PrusaSlicer/` folder on Linux, to no avail.

#### Project File (.3MF) where problem occurs

Any project.

rtyr, 2020-06-30:
Configuration bundle 1.1.2 was bundled with PrusaSlicer 2.2.0 for all platforms. Newer bundle versions are downloaded/updated on-line. Check the "Update built-in Presets automatically" parameter in PrusaSlicer preferences.

I am not Linux user, but I tried it on Ubuntu 20.04/appimage and bundle update worked fine.

[image]

goncalossilva, 2020-06-30:
Thanks @rtyr. I did check this before but forgot to include it. It's pretty vital for the report.

Unfortunately, I do have that enabled on both platforms, to no avail. Manual check fails too.

[image]
[image]

goncalossilva, 2020-06-30:
When I start PrusaSlicer from the terminal, it outputs this:

```
[2020-06-30 18:04:46.708481] [0x00007f279a62b700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]
[2020-06-30 18:04:46.894398] [0x00007f279a62b700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]

```

It seems related as that file lists the various configurations.

bubnikv, 2020-07-20:
SSL certificates are stored at different locations on different Linux distros. Linux users are shooting themselves into their legs again.

@tamasmeszaros implemented a solution for PrusaSlicer already, it will be in PrusaSlicer 2.3.

jhoblitt, 2020-09-24:
I also just discovered that I didn't update to 1.1.9 and the K values I'm getting are garbage under fw 3.9.1:

```
[2020-09-24 16:18:26.086637] [0x00007ff140bbb700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]
[2020-09-24 16:18:26.694025] [0x00007ff140bbb700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]
```

But I am able to use use curl and wget without geting a tls error:

```
$ curl -sSL -I http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch/index.idx
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Thu, 24 Sep 2020 23:21:52 GMT
Content-Type: text/html; charset=iso-8859-1
Connection: keep-alive
Location: https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch/index.idx

HTTP/2 200 
server: nginx
date: Thu, 24 Sep 2020 23:21:52 GMT
content-length: 7702
last-modified: Thu, 24 Sep 2020 23:00:43 GMT
etag: "1e16-5b0172b6f8a39"
accept-ranges: bytes
```

jhoblitt, 2020-09-24:
Kludge for fedora/el:

```
 /etc/ssl/certs $ sudo ln -s ca-bundle.crt ca-certificates.crt
```

bubnikv, 2020-10-09:
Fixed with 
1123689a226b243c15ca0748ebb5f9f029abe830
611a2434479fe0a9b20426eeb323ce15d94f506c
by searching for system SSL certificates at various system locations.

It will be fixed in upcoming PrusaSlicer 2.3.0-alpha1.

closed 2020-10-09 by bubnikv