prusa3d/PrusaSlicer #4449: Feature Request: Add a 'Prime initial extruder' option.

BoyStot, 2020-06-28:
With single extruder multi material printers It would be useful to have a Prime initial extruder option so that the printer does not need to swap through all of the colours at the start when you only need it to purge out the starting extruder.

At the moment you either need to 
Prime all printing extruders which is a waste of time and material for all but the initial extruder
Manually purge by extruding filament manually before printing.

If you use the second option you also need to add filament load gcode to the start of the script.

Having the Prime Initial Extruder option would allow this to be taken care of automatically.

lukasmatena, 2020-06-28:
Third option is to enable skirt, possibly with 'Minimal filament extrusion length' set to a non-zero value. Is that not enough for some reason?

BoyStot, 2020-06-29:
I can try that.  Depending on the size of item I cant always get a skirt as I'm printing close to the perimeter already and adding a skirt or brim makes it too big to fit.

rtyr, 2020-06-30:
You can also set your own custom priming (purge line, etc.) using start g-code and `[initial_tool]` placeholder. 

T[initial_tool]

rtyr, 2020-07-11:
Can we close this issue?

rtyr, 2020-07-26:
No response. Closing.

closed 2020-07-26 by rtyr