prusa3d/PrusaSlicer #4453: Extrusion Width Too Small

Balance2, 2020-06-29:
### Version
PrusaSlicer 2.2.0 64bit

### Operating system type + version
Windows 10 Enterprise 10.0.17763 and Windows 10 Pro Version 10.0.18363

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S with/without MMU

### Behavior
Error Encountered when slicing.
[image]

_Is this a new feature request?_No

#### Project File (.3MF) where problem occurs
[CamMount Test1.zip](https://github.com/prusa3d/PrusaSlicer/files/4847094/CamMount.Test1.zip)

The goal here is to print an object a specified distance above the print bed in order to insert a fastener into the print.  A raft fills in the space where the fastener shaft goes so it isn't an option.  While the slice visually looks acceptable it will not export and the error in the screenshot above appears.

Thanks!

Balance2, 2020-07-21:
Hello!  I have verified that previous projects have worked just fine but the latest update to Prusaslicer has broken the ability to print on a custom raft.  The error I get from my previous design that printed using this method is below.

[image]

I also attached the project as well.

[Bolt Head.zip](https://github.com/prusa3d/PrusaSlicer/files/4954330/Bolt.Head.zip)

n8bot, 2020-07-21:
There were commits yesterday and today that might fix the empty layers issue. Can you try building from latest master and seeing if it works now?

Balance2, 2020-07-21:
I can do that.  I did find that the problem can be circumvented by intersecting the layer with a support enforcer and turning on supports on enforcers only.

lukasmatena, 2020-07-21:
@Balance2 Your 3MF reports problem at 18.4, not at 18.5. And the cause is that it detects that the two layers (18.4 and 18.0) are too far apart (given current layer height). It is 18.2 that is actually empty.

The solution is coming, as mentioned by @n8bot already. The error will be converted into a warning (for all layers but the first), so the users can do the decision about whether it is ok to print themselves. It should already be in current master (except that the warning is not shown yet).

Balance2, 2020-07-30:
So I have an update on this.  The missing layer, 18.2, is the bottom layer of the part.  It's not printing the bottom layer at all so directly above the custom raft, which has a top, is a gap and then directly to infill supported by supports generated using support enforcers but no bottom layer at all.  Prior to the update this exact 3mf printed without support enforcers and the bottom layer of the upper part printed as one would expect.  So asking me if I don't mind the missing layer isn't going to solve the problem that the layer shouldn't be missing.

lukasmatena, 2020-08-05:
@Balance The layer is missing because there is a gap between the objects. Your layer height is 0.2mm. When slicer creates layer 18.2 (the number is the top of the layer), it slices the scene at 18.1mm (print_z-0.5*layer_height). That's just below the top part, so is does not see anything on that layer. The slicing is always discrete and slicer can hardly do anything else. Move the top part lower if you don't like it. Why do the parts have to be on top of each other anyway?

Balance2, 2020-08-05:
First, I want to thank you all for helping out.  Each post yall have made has made me re-examine or dig deeper and I have found a lot here that I am about to go into.  Anyways thanks again!

Okay so the pictures above give you an idea of what I am doing.  The part on the right is a knob that has a fastener embedded in it.  In order to embed the fastener I need to raise the knob up so the threaded portion of the fastener can go down or else the print head will hid the threaded shank.  To do this you create a another part to raise the primary part up in space because Prusaslicer won't do that for you.  This is the first version of a raised embedded fastener I ever made.

The original part was a similar thing.  You take a disk and you make it an support blocker that also provides an intentional gap between the part and the print bed.  Both of those .3mf files worked just fine on my prior version but that was PrusaSlicer 2.0.0.  Here is where things were going sideways.  

In 2.2.0 I get the gap error.  I trouble shot it at the time by changing lots of settings to see what happened and in doing so I changed the print setting from .20mm Quality to .15mm Quality(here I found a different problem I just didn't notice it yet).  I found a work around using support enforces for the gap and the prompt to ignore will be nice.  In effect he original issue is resolved.

I then went to print the knobs and found I had no bottom layer.  I assumed this was due to the support enforcer or the error and started testing lukasmatena's suggestion to lower or remove the gap but that did nothing.  I then tried PusaSlicer 2.0.0 and the .3mf I have on my hard drive now doesn't work anymore.  So I downloaded the one I uploaded here and found the layer height change from .20mm to .15mm is what was causing the bottom layers to go missing.  I tested every layer height preset from Prusa that I have and .15mm Quality and .15mm Speed both end up with missing bottom layers from the knob while all other presets include the bottom layers as intended.

So should I open a new issue since the original one is resolved and the new one is just about the .15mm layer height missing bottom layers in a very specific set of circumstances?

closed 2026-09-29 by prusaslicer-bot[bot]