prusa3d/PrusaSlicer #4455: Per layer infill pattern

reddawgesq, 2020-06-29:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mk3/MK3S/Mini, all with current F/W

New Request

I suggest adding a feature that will enable specifying a mid-print infill pattern that differs from either the normal, bottom layer or top layer patterns. The attached picture illustrates the reason for this request. It shows one part of a mold that I created. This part is joined to a second part (not pictured) and filled with a pourable RTV silicone elastomer to create an electric motor vibration isolator. It is necessary for the "top" layer to have a surface characteristic  as smooth as possible in order for there to be no gaps between the two parts of the mold, which could lead to "leaking" of the elastomer between them. However, for cosmetic reasons I would like the top surface of the depressed "cavity" to have the Hilbert Curve (or another decorative) pattern. I'm envisioning a process that,  after the initial slicing, you navigate to a specific layer, in much the same way you can now insert a "Pause" command, or "cut" the piece into top and bottom sections, and once at that layer, enable a context/drop-down menu with infill pattern selection effective for that layer only.
[image]

n8bot, 2020-06-29:
I believe this functionality is already present in current builds of PrusaSlicer.

At thr right sidebar, where the model is listed, there is a column that says 'Editing' and their is an icon you can click to get a menu. There is an option 'Height range Modifier' that allows you to change many settings for a specific range of height. Infill pattern can be specified.

I have not personally tested this functionality, but it's there.

reddawgesq, 2020-06-29:
Thanks, I will look into that.

n8bot, 2020-06-29:
Once you add the Height range Modifier, you'll have to click a new icon that appears beside that, and choose "Add settings->Infill" and then select 'Top fill pattern' from the drop list.

rtyr, 2020-06-29:
Yes, it can be done quite easily.

[image]

reddawgesq, 2020-06-29:
Excellent responses, thanks to all.

closed 2020-06-29 by reddawgesq