prusa3d/PrusaSlicer #4457: Incorrect size/volume shown in Info pane after scale factor change

paynterf, 2020-06-30:
### Version
2.2.0+win64

### Operating system type + version
Win 10 64 bot
### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
The Size and Volume numbers don't change when the scale factors are changed

* _Steps needed to reproduce the problem_
 Put any model on the plater. Change the scale factor from 100% to anything else

* _Expected Results_
The size and volume in the Info pane should update to reflect the new scale factor(s)

* _Actual Results_
Nothing changes - the size and volume numbers remain the same

rtyr, 2020-06-30:
I was able to reproduce it. Size info is updated to proper values after deselecting/selecting object, volume stays unchanged.

ghorn, 2021-07-16:
This is still incorrect in the 2.3.2 release.

closed 2026-09-29 by prusaslicer-bot[bot]