prusa3d/PrusaSlicer #4458: View mode reverts to "tool".

MarcoAlici, 2020-06-30:
### Version
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 (build 19041), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Every time (except the first one) I press "slice now" button, the view mode is reverted to "tool", no matter what view mode is selected. (I usually want to have it set to "Feature type").

[image]
[image]
[image]

Vonnye, 2020-12-19:
Also discussed in Prusa Forum (https://forum.prusaprinters.org/forum/prusaslicer/change-default-settings-in-preview/) where Alicia provides a patch to retain the previous view ( https://ion.nu/tmp/prusaslicer_rememberpreview.patch).

Kachidoki2807, 2020-12-22:
This is solved since 2.3.0-beta1

> Logic of automatic selection of View mode in Preview (Feature Type/Tool/etc.) was slightly reworked so it does not switch the visualization type when not needed. Namely, when a custom tool change was present, the mode would annoyingly switch to 'ColorPrint' during any parameter change. (#5080)

lukasmatena, 2020-12-22:
@MarcoAlici Can you please retest on your system in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2)?

@Kachidoki2807 Thanks for trying to help make the issue tracker ~perfectly tidy~ slightly less messy.

MarcoAlici, 2020-12-23:
> @MarcoAlici Can you please retest on your system in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2)?

It works!
And yes, as stated by @alicia  [here](https://forum.prusaprinters.org/forum/prusaslicer/change-default-settings-in-preview/) the issue happened when the number of extruders is set to a number greater than 1.

Thank you all!!!

lukasmatena, 2020-12-23:
Thanks. Closing the issue.

closed 2020-12-23 by lukasmatena