prusa3d/PrusaSlicer #4460: Bridge flow ratio affecting solid infill erroneously

timothyhibbard, 2020-06-30:
### Version
_2.2.0+_

### Operating system type + version
macOS Catalina Version 10.15.5

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S

### Behavior
When adjusting the Print Settings > Advanced> Flow > "Bridge flow ratio" setting from the default (.95) to about half (.5 or less) the second and third layers, which are supposed to be solid infill layers, change to bridge infill layers. this is causing my second and third layers to not stick together properly as shown here: 
[image]

This is a picture of the second layer with bridge flow at .95:
[image]

This is a picture of the second layer with bridge flow at .5:
[image]

The expected results are that any adjustment to bridging parameters only affect layers that are actually bridges. I believe the bridge flow ratio parameter is affecting solid infill layers erroneously. When the bridge flow ratio is decreased, it causes a small gap between the first and second layer and thus the slicer thinks it's a bridge, when it is not.

#### Project File (.3MF) where problem occurs

[wrong.infill.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4854499/wrong.infill.3mf.zip)

wrong.infill.3mf

rtyr, 2020-07-01:
May be related to https://github.com/prusa3d/PrusaSlicer/issues/3037.

timothyhibbard, 2020-07-01:
> May be related to #3037.

rtyr, yes, I believe you're right that it is related. Unfortunately, they do not solve the problem.

Gerstacker, 2020-11-17:
There's an object I'd like to print with BFR=0.4, but changing from 0.5 to 0.4 turns the top solid infill into bridges, blowing a 5-hour print to 17 hours due to bridging speed.
Logically, it doesn't make sense that bridge flow ratio would contribute to the decision about whether an extrusion is a bridge in the first place or not.

On a tangent, I don't understand why it's a ratio in the first place. With a given nozzle, sometimes you might print a line width of 133% of nozzle diameter and sometimes 150% depending on global properties of the model that don't have anything to do with bridging. If I switch perimeters from 133% to 150%, why should bridge flow increase at all?

closed 2026-09-29 by prusaslicer-bot[bot]