prusa3d/PrusaSlicer #4461: Feature request: Per part fan speed

drpain, 2020-07-01:
It would be awesome if you were able to set the fan speed per model on the build plate. 

Some items require additional fan speed due to complexity and would be awesome to be able to set those speeds per part

Sandiegoo, 2020-07-30:
it would be also nice to have a setting where you can set the fan speed for each individual layer. For some complex parts this could be helpful

neophyl, 2020-07-30:
Adding fan speed to the list of available modifiers would be handy.  That way you could do it for specific areas of a model too.

As a workaround if you want to adjust the fan speed on a per layer basis you could use the add custom gcode at layer facility.  Just slice it as normal, adjust the layer slider on the preview to the layer you require and then right click the + icon on the slider and select add custom gcode.  You can then add the fan speed in there using M106 Sxx.  Just remember that the speed will be set to that until its changed elsewhere.

well-engineered, 2020-12-21:
Also, there is no way (I know of) to set individual fan speeds for different Tools in a multi-tool printer

PhilBaz, 2025-01-23:
Yes, another reason this is important is when you build custom supports....Especially for PETG.

I've built large custom support objects, but there is no way to have the top surface receive 100% fan to ensure easy removal. 

Please consider!

closed 2026-09-29 by prusaslicer-bot[bot]