prusa3d/PrusaSlicer #4463: [Feature Request] filament sensor homing after filament change

Quaziz-creator, 2020-07-01:
It would be nice to have gcode inserted after a pause or filament runout, that homes the printer in x and y so that it can resume from where it was printing before, even if one has moved the axis by mistake. I often do this while changeing the filament (in the frustrating stock extruder Creality puts on the printer, no matter if it is premium or not) or fiddeling with the printer while in paus. I have tried to reach out to Creality about this, but have not got any answer, if they are going to do something about it. If i was a rich guy I would have bought a prusa printer instead.

rtyr, 2020-07-01:
Behavior during filament runout/pause must be configured in printer firmware. You have to ask Creality (as you did) or use alternative firmware. This is not a task for a slicer.

Quaziz-creator, 2020-07-01:
Thanks for the answer. Yeah, that's true. You can't capture an event and act on it i g-code, that true. Do the prusa printers supports such features, out of the box?

rtyr, 2020-07-01:
Yes. For example, MK3S moves print head to the right-front corner for filament change, and to left-back corner when paused (so printed object is easily accessible). There is no risk of moving with the print head accidentally because motors are still powered. We are using our custom firmware (based on Marlin) and our firmware team constantly works on updates and new features.

rtyr, 2020-07-01:
Thank you. You can see some upcoming features here https://www.youtube.com/watch?v=NAW2JR-D_QU (from roughly 25:00) :-). I believe we can close this issue.

Quaziz-creator, 2020-07-01:
Yes, we can! Thank you very much!

closed 2020-07-01 by Quaziz-creator