prusa3d/PrusaSlicer #4474: [Feature Request] Monotonous on all Top and Bottom layers, not just the first

Area5142, 2020-07-04:
### Version
PrusaSlicer Version:   version_2.3.0-alpha0+linux-x64
Build:     Commit: d8a81e978a8e95d16ec269a77be37dabda6cccff, compiled from source on Linux

### Operating system type + version
Linux Mint 19.3 x64

### Behavior
If Monotonous is selected the first layer is correctly printed in one direction and next layers is printed in "old style" rectilinear.
This works very well for opaque filament, but when using transparent filaments the shift in style can be seen through the first layer.

### Expected behaviour
When Monotonous is selected, all layers is printed in that style, not only the first. This shouldn't have an impact on printing time, but will look better when using transparent filaments.

BTW: Monotonous is a really great feature!

n8bot, 2020-07-04:
Agreed. This issue was already raised here: #4425 Monotonous infill for middle solid fill layers

rtyr, 2020-07-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4425. Closing.

closed 2020-07-04 by rtyr

timur-tabi, 2020-10-17:
For the record, "monotonous" is not the right word. The correct word is "monotonic".

rtyr, 2020-10-17:
We are aware of that.

[image]